e-m-b-a/emba #22: I get a error when i run the install.sh script

clb-by, 2020-12-17:
the error is these
[image]

i install the emba successfully before, but this i can't install successful

clb-by, 2020-12-18:
ok! I have resolved！

clb-by, 2020-12-18:
binutils don't have include the <limits.h> and <fcntl.h> successfully，i reimport <limits.h> and <fcntl.h> head files

p4cx, 2020-12-18:
Thank you for your research. I will update the installer soon, with some improvements overall. There I will update the binutils version to  2.35.1 instead of 2.34. Maybe they fixed it themselves and imported the right files.

clb-by, 2020-12-18:
Maybe！I think this tool is pretty good!

closed 2020-12-18 by clb-by