e-m-b-a/emba #40: Cleanup output of s10 - vul_func_basic_check function

m-1-k-3, 2021-01-22:
The output of objdump is misaligned:

[image]

p4cx, 2021-03-10:
How do you then imagine it, for example, when something like that appears?

```
    00421a30	DF	*UND*	00000000	strcpy
    004219c0	DF	*UND*	00000000	printf
    004218f0	DF	*UND*	00000000	sprintf
    004218e0	DF	*UND*	00000000	vsnprintf
    004218c0	DF	*UND*	00000000	fprintf
    0041d4b4	g	DF	.text 	00000074	Base	lxmldbc_system
    00402d94	g	DF	.text 	00000084	Base	client_printf
    00421710	DF	*UND*	00000000	snprintf
    00421600	DF	*UND*	00000000	strcat
    004215d0	DF	*UND*	00000000	vfprintf
    00421580	DF	*UND*	00000000	system

```

m-1-k-3, 2021-03-10:
Currently we get a huge amount of uninteresting stuff. We have to only search for the functions we have in our config files and then the alignment is probably automatically right.

p4cx, 2021-03-11:
Is a small fix, only added -w to one grep command in S10 - will add this fix to #65

closed 2021-03-15 by m-1-k-3