e-m-b-a/emba #59: Question: What OSes are supported?

fkromer, 2021-03-01:
Thanks a lot for this great project. I fast forwarded through the wiki but was not able to figure out which embedded Linux operating systems are supported. Probably some distributions are not supported at all. Probably some checks are meant for specific distributions only. There are so many Linux-based embedded OSes like e.g. [Android Things](https://developer.android.com/things), [Android Cars](https://developer.android.com/cars), [OpenWRT](https://openwrt.org/), [Yocto Project](https://www.yoctoproject.org/) embedded Linux, [Ubuntu Core](https://ubuntu.com/core), [uClinux](https://en.wikipedia.org/wiki/ΜClinux), ... Could you provide more info about distro compatibility?

p4cx, 2021-03-01:
Thank you, we hope you enjoy it. The simple reason you couldn't find a list of distributions is because emba doesn't focus on these, it focuses on Linux in general. Many distributions share the basic file structure, and this is how we do our checks. Also, in many places we are only looking for specific patterns, and if they are in a different place we will find them too. More specific tests for individual distributions are a good idea for the future development of emba, but are currently hardly represented.

In addition to Linux, emba can also recognize RTOS systems, for example, and then search it for static version numbers.

fkromer, 2021-03-02:
@p4cx So far I've not even played around with it. But in this case I probably will do so. I just wondered because e.g. Ubuntu Core differs from other embedded Linuxes significantly cause of it's sandboxing concept (snap apps). Consequently there are several rootfs, one for every app. Cool, that the tool should work with Zephyr as well then.

p4cx, 2021-03-03:
As far as I know, Ubuntu Core is built in exactly the same way as a classic Linux. All configurations in the / etc directory and so on.At the moment we're developing Emba for our needs and we didn't have an Ubuntu core image yet so we didn't create specific tests for it.
Try it out and tell us about your experience. If you want, you can add some functionality to emba for Ubuntu Core.

closed 2021-03-08 by p4cx