e-m-b-a/emba #100: #100 - the future of emba

m-1-k-3, 2021-04-24:
What are you missing in emba. Let us know how you are using emba, what is working and what doesn't work

From our point of view we have the following areas to improve:
* improve threading (more modules should use it)
* improve the web reporter
* emba needs to get faster again (we lost a lot of performance with this pull request https://github.com/e-m-b-a/emba/pull/92)
* some modules need some cleanup (e.g. P07)
* switch from docker to podman

What else would be great to bring to emba?

m-1-k-3, 2021-09-01:
all done.

closed 2021-09-01 by m-1-k-3

dgutson, 2024-09-24:
@m-1-k-3 what happened with "switch from docker to podman"?

m-1-k-3, 2024-09-24:
@dgutson not enough time to go into the depths ...

dgutson, 2024-09-24:
@m-1-k-3 would you accept such contribution, or it has been discarded?

m-1-k-3, 2024-09-25:
I would prefer to open an Issue for tracking first. We need to define some generic testcases/testfirmware and for some special modules some dedicated firmware images. With this in place we can walk through all modules and check if they are working as expected with podman and we can document it there in a simple list like:

- [ ] emba startup - testfirmware used - works [y/n]
- [ ] Module P02 - testfirmware used - works [y/n] - comment 
- [ ] Module P05 - testfirmware used - works [y/n] - comment
- [ ] ...

After evaluating what is working and what not we can think on fixing and how to fix. If you take this over and start up with some initial testing and documenting it would be total awesome.