e-m-b-a/emba #111: Ideas for the web reporter

m-1-k-3, 2021-05-21:
@p4cx has completely rewritten the emba html reporter here: https://github.com/e-m-b-a/emba/pull/101

With this base we now have great possibilities. This issue is a collection and discussion of some ideas:

- [ ] Linking to submodules from aggregator eg. 
"[+] 1091 files and 91 directories detected." should link to "html-report/s05_firmware_details.html"
"[+] Found 182 issues in 78 shell scripts." should link to "html-report/s20_shell_check.html"
- [ ] Linking to vulnerability details from aggregator
"[+] Found version details: 	dnsmasq             	:	2.45           	:	CVEs: 21      	:	Exploits: 8       " should link to the details of dnsmasq within "html-report/f19_cve_aggregator.html"
- [ ] the current visualized firmware file is not included
- [ ] Inter-module linking eg.:
- [ ] "[+] Found the following root directory: ../emba_logs/logs_dir300/firmware/DIR300B5_FW214WWB01.bin_binwalk_emba/_DIR300B5_FW214WWB01.bin.extracted/squashfs-root" could link to "html-report/s05_firmware_details.html"

p4cx, 2021-05-27:
Actual work progress in [webreport_updates](/p4cx/emba/tree/webreport_updates):

- [ ] Linking to submodules from aggregator eg. 
   - [ ] "[+] 1091 files and 91 directories detected." should link to "html-report/s05_firmware_details.html"
   - [ ] "[+] Found 182 issues in 78 shell scripts." should link to "html-report/s20_shell_check.html"
   - [ ] Kernel version can link to the kernel module
- [ ] Linking to vulnerability details from aggregator
"[+] Found version details: 	dnsmasq             	:	2.45           	:	CVEs: 21      	:	Exploits: 8       " should link to the details of dnsmasq within "html-report/f19_cve_aggregator.html"
- [ ] the current visualized firmware file is not included
- [ ] Inter-module linking eg.:
"[+] Found the following root directory: ../emba_logs/logs_dir300/firmware/DIR300B5_FW214WWB01.bin_binwalk_emba/_DIR300B5_FW214WWB01.bin.extracted/squashfs-root" could link to "html-report/s05_firmware_details.html"
- [ ] links to further details
  - [ ] shellchecker results
  - [ ] deep search results
  - [ ] php results
- [ ] include emba.log into web reporter
- [ ] Parallelize creation of info files (like shellchecker, deep search, php results)

p4cx, 2021-06-15:
The PR #121 now has most of the entries of this feature list. 

Definitely missing:
Parallelization, because I tried it multiple time and **everytime** it was slower than leaving it, how it was. 
Some links/anchors - those can be added in the future and of each contributor. 

After merging #121 this issue #111 will be closed.

m-1-k-3, 2021-06-15:
merged and closed ...

closed 2021-06-15 by m-1-k-3