e-m-b-a/emba #126: Sed Error

seanog8, 2021-06-20:
I have the Error message which you can see in the screenshot. Im not sure if it came with my changes of my fork. If I remember right it is also appearing in the master branch. It is only in the console output, but not in the logfile.

[image]

seanog8, 2021-06-20:
Maybe you can check next time you run a scan if it is appearing for you as well. Otherwise I'll recheck it with a firmware-scan with the master branch soon.

m-1-k-3, 2021-06-21:
Thanks for reporting. Have you activated the web-reporter? If so, could you please give it a run without the web reporter and check if the error also appears?

p4cx, 2021-06-21:
Looks like a bug in the webreport (position wise) - I will check this in near future.

seanog8, 2021-06-21:
Yes, I ran a small scan without webreport. It is not appearing there.

PR #132 (2021-07-02): Fix webreport and other stuff
Fix #126 and fix #129 and improve speed on web report.
closed 2021-07-02 by m-1-k-3