e-m-b-a/emba #174: Files not found by 50_authentication_check

aspi-dwtt, 2021-11-05:
From the logfile, I conclude that certain files in /etc where not found by the tool. Am I correct?
These files are present in the tested firmware image!

```
[+] Check users, groups and authentication
=================================================================

==> Users with UID zero (0)
-----------------------------------------------------------------
[*] Searching accounts with UID 0
[-] /etc/passwd not available

==> Non-unique accounts
-----------------------------------------------------------------
[*] Searching non-unique accounts
[-] /etc/passwd not available

==> Unique group IDs
-----------------------------------------------------------------
[*] Searching non-unique group ID's
[-] /etc/group not available

==> Unique group name
-----------------------------------------------------------------
[*] Searching non-unique group names
[-] /etc/group not available

==> Query user accounts
-----------------------------------------------------------------
[*] Reading system users
[-] /etc/passwd not available

==> Query NIS and NIS+ authentication support
-----------------------------------------------------------------
[*] Check nsswitch.conf
[-] /etc/nsswitch.conf not available
```

m-1-k-3, 2021-11-05:
Interesting ... could you provide a link to the firmware?

aspi-dwtt, 2021-11-05:
I'm sorry I can't share the firmware with you. 
But I'm happy to test again when you've found a solution (a test run takes about 11 hours to complete in my current test setup)

p4cx, 2021-11-05:
That makes it quite complicated to fix the bug, because the code should give this error only if there is no file. Maybe you can add some `echo` statements in module *S50* to give us a more verbose output, e.g. between line 92 and 93: 
``` bash
  for PASSWD_FILE in "${PASSWD_FILE_PATHS[@]}"; do
    echo "$PASSWD_FILE"
    if [[ -f "$PASSWD_FILE" ]] ; then
```

To only run the module S50, you can specify this in the run command:
``` bash
./emba.sh -f path_to_firmware -l path_to_log -W -t -r -s -m S50
```

m-1-k-3, 2021-11-05:
ist there a passwd file? Is this file in /etc or is it somewhere else?

aspi-dwtt, 2021-11-05:
Hello Pascal,

I’ve made the adaption you requested, and started a test run.
I had to start the command with sudo.

I’ve have to leave the office now, I will get back to  you on Monday

Have a nice weekend.

Alberto Spin
Product Designer

D +31 88 0032 951
P +31 88 0032 900
M +31 6 52 51 62 51

Datawatt B.V.
Woldmeentherand 5
8332 JE Steenwijk
The Netherlands
[Ovarro Logo]<https://www.datawatt.nl/>

    [Datawatt] <https://www.datawatt.nl/>        [Datawatt  Twitter] <https://twitter.com/datawatt>        [Datawatt LinkedIn] <https://www.linkedin.com/company/datawatt-b.v./>

DATAWATT B.V., a private company with limited liability (besloten vennootschap met beperkte aansprakelijkheid), having its seat (zetel) in Steenwijk and its address at
Woldmeentherand 5, 8332 JE Steenwijk, registered with the Trade Register under number 01042647. Although Ovarro has taken reasonable precautions to ensure
no viruses are present in this email, Ovarro cannot accept responsibility for any loss or damage sustained as a result of computer viruses and the recipient must ensure
that the email (and attachments) are virus free. This email including any attachments is for the named person or persons use only. It may contain confidential or legally
privileged information. No confidentiality or privilege is waived or lost by any mis-transmission. If you receive this email in error, please notify the sender, delete the email
immediately, and remove all copies from your system. You must not, directly or indirectly, use, disclose, distribute, print, or copy any part of this message if you are not
the intended recipient. All opinions expressed in this message are that of the individual sender, except where the message states otherwise and the sender is so authorised.
Any liability (in negligence or otherwise) arising from any third party relying on this email is excluded. Ovarro reserves the right to monitor all email communication for
compliance purposes and to protect our business. To view our Privacy Policy, click here<https://www.ovarro.com/privacy-policy/>.

Please consider the environment and do not print this email unless absolutely necessary. Ovarro encourages environmental awareness.

aspi-dwtt, 2021-11-05:
Hello Michael,

Yes,

the passwd, group, shadow and gshadow all all located in /etc

Alberto Spin
Product Designer

D +31 88 0032 951
P +31 88 0032 900
M +31 6 52 51 62 51

Datawatt B.V.
Woldmeentherand 5
8332 JE Steenwijk
The Netherlands
[Ovarro Logo]<https://www.datawatt.nl/>

    [Datawatt] <https://www.datawatt.nl/>        [Datawatt  Twitter] <https://twitter.com/datawatt>        [Datawatt LinkedIn] <https://www.linkedin.com/company/datawatt-b.v./>

DATAWATT B.V., a private company with limited liability (besloten vennootschap met beperkte aansprakelijkheid), having its seat (zetel) in Steenwijk and its address at
Woldmeentherand 5, 8332 JE Steenwijk, registered with the Trade Register under number 01042647. Although Ovarro has taken reasonable precautions to ensure
no viruses are present in this email, Ovarro cannot accept responsibility for any loss or damage sustained as a result of computer viruses and the recipient must ensure
that the email (and attachments) are virus free. This email including any attachments is for the named person or persons use only. It may contain confidential or legally
privileged information. No confidentiality or privilege is waived or lost by any mis-transmission. If you receive this email in error, please notify the sender, delete the email
immediately, and remove all copies from your system. You must not, directly or indirectly, use, disclose, distribute, print, or copy any part of this message if you are not
the intended recipient. All opinions expressed in this message are that of the individual sender, except where the message states otherwise and the sender is so authorised.
Any liability (in negligence or otherwise) arising from any third party relying on this email is excluded. Ovarro reserves the right to monitor all email communication for
compliance purposes and to protect our business. To view our Privacy Policy, click here<https://www.ovarro.com/privacy-policy/>.

Please consider the environment and do not print this email unless absolutely necessary. Ovarro encourages environmental awareness.

aspi-dwtt, 2021-11-08:
Console output from the run with the extra 'echo' statement:

`
[+] Check users, groups and authentication
=================================================================
..
==> Users with UID zero (0)
-----------------------------------------------------------------
...[*] Searching accounts with UID 0
..........
...[-] /etc/passwd not available
................
`

PR #176 (2021-11-08): Remove broken code
Fix #174 - there is a check, if the files are valid and it was broken. But we are checking in the modules, if the paths are valid (`if -d` or `-f`). Remove it, now it works :)
closed 2021-11-08 by m-1-k-3