e-m-b-a/emba #175: License details missing

m-1-k-3, 2021-11-05:
We have started to include license details in our version identifier configuration here: https://github.com/e-m-b-a/emba/blob/master/config/bin_version_strings.cfg

This is in a very early stage and need your help. If you know the license of some of the software components please include it directly in the configuration and bring up a pull request or comment to this issue with the license and the source for verification.

doboianh, 2021-11-05:
can i help you ?

m-1-k-3, 2021-11-05:
For sure. If possible fork our repo and include missing license details in https://github.com/e-m-b-a/emba/blob/master/config/bin_version_strings.cfg
Afterwards please make a pull request and include links to the license details in the PR details.

mucahitdemir, 2022-01-06:
so what is the problem there?

m-1-k-3, 2022-01-07:
Hi @mucahitdemir, there is not a problem. Someone needs to do it ;) 
In the mentioned configuration file are a lot of "unknown" entries. For correct license identification we need for every software component the correct license included here. If it is something special it needs also included into the license link file here: https://github.com/e-m-b-a/emba/blob/master/config/bin_version_strings_links.cfg

mucahitdemir, 2022-01-07:
Hi again, you're right.
Sincerely

m-1-k-3, 2022-04-19:
Looks not too bad anymore ... will close this for now

closed 2022-04-19 by m-1-k-3