e-m-b-a/emba #177: net-tools installation on Ubuntu failed

m-1-k-3, 2021-11-10:
I'm running in a fresh Ubuntu VM and cloned EMBA from git then ran the installer and yes, it seems to have failed to install the net-tools.

I *could* (untested!) imagine that the way you handle parameters with a dash (-) in it, that it somehow breaks because the dash is interpreted as a command line parameter switch. But that's just wild speculation.

_Originally posted by @floyd-fuh in https://github.com/e-m-b-a/emba/issues/118#issuecomment-965539783_

m-1-k-3, 2021-11-10:
@floyd-fuh I opened an issue for this. Which installation mode are you using?

floyd-fuh, 2021-11-11:
Exactly what the wiki told me, `sudo ./installer.sh -d`

m-1-k-3, 2021-11-11:
I think I found the problem ... could you do a quick check with my current dev fork: https://github.com/m-1-k-3/emba

m-1-k-3, 2021-11-11:
Regarding the other topic: EMBA is a docker container. With the installation -d you download the EMBA docker and install only CVE search on the host

m-1-k-3, 2021-11-13:
fixed

closed 2021-11-13 by m-1-k-3