e-m-b-a/emba #187: cve-search issues after latest updates

m-1-k-3, 2021-11-26:
With the latest updates we have isolated the EMBA container from a networking perspective. This means EMBA is not able to reach the internet anymore. With this also some adjustments in the EMBA docker container and in the local mongodb configuration were needed.

The easiest way to get an updated EMBA installation are the following steps:
```
cd emba-directory
git pull
sudo docker pull embeddedanalyzer/emba
sudo ./installer.sh -d
```

If you are running into issues that the EMBA container is not able to access your mongodb installation you can check the following:

* ifconfig should show a new network interface with the following ip address:
```
└─$ ifconfig emba_runs
emba_runs: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.36.0.1  netmask 255.255.0.0  broadcast 172.36.255.255
        ether 02:42:cb:c9:37:80  txqueuelen 0  (Ethernet)
        RX packets 15643  bytes 964808 (942.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 43695  bytes 1435782652 (1.3 GiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
If this is not the case you can try to restart docker at all with the following command
`└─$ sudo /etc/init.d/docker restart  
`
Another way is to delete the network interface and re-initialize it:
```
└─$ sudo docker network rm emba_runs
└─$ sudo docker-compose up --no-start
```

* Your mongod.conf should have the following bindIp configuration:

```
└─$ grep bindIp /etc/mongod.conf
  bindIp: 172.36.0.1

```

* Finally you can test it manually via executing EMBA and test the cve-search connection the following way:
```
└─$ sudo docker-compose run emba     
WARNING: The FIRMWARE variable is not set. Defaulting to a blank string.
WARNING: The LOG variable is not set. Defaulting to a blank string.
WARNING: The EMBA variable is not set. Defaulting to a blank string.
Creating emba_forked_emba_run ... done
┌──(root@d2a5960b73b2)-[/emba]
└─# /external/cve-search/bin/search.py -p busybox
```
This should show some CVE details regarding busybox.

closed 2022-07-16 by github-actions[bot]

Hellworold, 2022-09-29:
Hi,
When iam trying to run the utility it gives an error CVE- search not working , check issue number 187 thus iam here .
Now i have checked all the steps above.
for some reason this is the output 
i have checked with some older versions 

[image]

Ret2c7, 2023-05-06:
Hi,
When iam trying to excute ./external/cve-search/sbin/db_updater.py -f，error have occurred。
[image]

m-1-k-3, 2023-05-06:
Please check the following:

* is your network ok? -> ifconfig should show the `emba_runs` network device with ip address `172.36.0.1`
* is mongod running? `netstat -anpt | grep 27017`
* is redis running? `netstat -anpt | grep 6379`

Ret2c7, 2023-05-06:
it seems like mongod didn't running 
[image]
[image]
[image]
[image]

m-1-k-3, 2023-05-06:
Then check mongod, restart it, check logs and as it is running you can do an update of the database.

Darren-zl, 2023-05-10:
I follow the above operations to do, are normal. But I still show cve-search -no ok. I don't know why.
[image]
[image]
[image]
[image]

Ret2c7, 2023-05-10:
try again after ./external/cve-search/sbin/db_updater.py -f

brainsht, 2023-05-10:
Found another source of cve-serach issues. Kept getting this error message while running ./external/cve-search/sbin/db_updater.py -f

`CVEDownloads - ERROR - Did not receive last-modified header in the response; setting to default (01-01-1970) and force update! Headers received: {'content-length': '93', 'cache-control': 'no-cache', 'content-type': 'text/html', 'connection': 'close'}`

@Ret2c7 try this one. Error is gone by adjusting max_workers to 1 in https://github.com/cve-search/cve-search/blob/1f0b50aa46814e2a683a9b7b01da8bcc0403154e/lib/DownloadHandler.py#L121

Afterwards ` ./external/cve-search/sbin/db_updater.py -f` works fine (takes longer though) and no cve-serach issue anymore for me. 

Really seems like there is a rate limit with https://nvd.nist.gov/
Source: https://github.com/cve-search/cve-search/issues/890#issue-1245042453

m-1-k-3, 2023-05-10:
@brainsht thank you for documenting this issue. I will include it into the main post.

aserper, 2023-09-12:
In my experience, trying to install EMBA on a server with a lot of cores (I have it running on a 36 core aws machine right now) causes the same issues with updating and populating the CVE database.  A workaround that I found to that issue is to run the emba installer solely by utilizing two cores. While I didn't really look into why it helps, I assume that the large number of cores opens more threads which causes some DDOS protection mechanism on the remote server to kick in (again, just a guess). This can be done by running the installer program like so:
`sudo taskset -c 0,1 ./installer.sh -d` which will run the installation script using only two cores. I know that you can go into the files and change the code but I'm lazy ;)

m-1-k-3, 2023-09-15:
With version 1.3.0 we switched to python environment. This results in the need of activating the python environment before doing cve-search tasks. 

On the host system it is possible to activate it the following way:
```
└─$ source ./external/emba_venv/bin/activate
```

In the docker container it is possible to activate it the following way:
```
└─$ source /external/emba_venv/bin/activate
```

m-1-k-3, 2023-11-21:
It is now recommend to setup a NIST API key
See also https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

brainstorm, 2023-11-23:
> 
> `thread_map(self.download_site, sites, desc="Downloading files", max_workers=1)`
> 
> 
> Really seems like there is a rate limit with https://nvd.nist.gov/ Source: [cve-search/cve-search#890 (comment)](https://github.com/cve-search/cve-search/issues/890#issue-1245042453)

That `max_workers` argument is now exposed as `$HOME/.cvexplore/.env` as `MAX_DOWNLOAD_WORKERS` and you can define it along with the `NVD_NIST_API_KEY` variable.

`sudo taskset -c 0,1 ./installer.sh -d` is not yet documented in the wiki and I suspect that there are still threading issues within EMBA, probably (see: https://github.com/e-m-b-a/emba/issues/908#issuecomment-1825000276).

m-1-k-3, 2023-11-24:
If you are running into cve-search issues please test https://github.com/e-m-b-a/emba/pull/913