e-m-b-a/emba #193: {Enhancement} HTML toggles for false positives

muddydev, 2021-12-07:
This is a tall order but would be nice for the roadmap 

In most cases. the discoveries for the CVEs don't actually affect the product. For example, if I'm running a kernel version that has 200 CVE's and 7 exploits. When I look at those findings I notice the CVE's are just a raw version analysis but if you dig down into the CVE it can say stuff like "If IPV6 is enabled" "IF the following flag is enabled in x config". IT would be nice to have the ability to go into the HTML report and maybe toggle stuff off that you know is a false positive. 

Kina like this project lets you do https://github.com/Guezone/SECMON. 

The toggling could let you generate an XML or something that logs the CVE's that you could apply to your next scan --fpxml

m-1-k-3, 2021-12-07:
Thanks for your feedback. We will evaluate Kina. The idea is great.

m-1-k-3, 2021-12-07:
As EMBA will not address this and the right area for this is EMBArk I will move this issue to the EMBArk area. See https://github.com/e-m-b-a/embark/issues/21

m-1-k-3, 2022-09-13:
Does it makes sense to load a CVE blacklist file from the config directory?

The idea is that you can for example place multiple CVE lists in the config directory and create a scan profile per cve list. This give you the possibility to collect the kernel CVEs into a file and ignore it in the future. Another possibility would be to generate CVE lists for Metasploit exploits and so you can generate a list for Metasploit exploits and so on.

As an initial implementation I will include a blacklist that will get checked for CVEs and the included CVEs are ignored from further processing.

closed 2022-09-22 by m-1-k-3