e-m-b-a/emba #194: Installation Failure

muddydev, 2021-12-07:
[0;35m[1mThe cve-search database will be downloaded and updated![0m

[0;35mCheck if the cve-search database is already installed.[0m

[0;35mcve-search database not ready.[0m

[0;35mThe installer is going to populate the database.[0m
Starting redis-server (via systemctl): redis-server.service.
Traceback (most recent call last):
  File "/opt/emba/external/cve-search/./sbin/db_mgmt_cpe_dictionary.py", line 27, in <module>
    from lib.Sources_process import CPEDownloads
  File "/opt/emba/external/cve-search/sbin/../lib/Sources_process.py", line 17, in <module>
    from lib.DatabaseLayer import (
  File "/opt/emba/external/cve-search/sbin/../lib/DatabaseLayer.py", line 35, in <module>
    mongo_version = db.command("buildinfo")["versionArray"]
  File "/usr/local/lib/python3.9/dist-packages/pymongo/database.py", line 757, in command
    with self.__client._socket_for_reads(
  File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/mongo_client.py", line 1387, in _socket_for_reads
    server = self._select_server(read_preference, session)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/mongo_client.py", line 1346, in _select_server
    server = topology.select_server(server_selector)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 244, in select_server
    return random.choice(self.select_servers(selector,
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 202, in select_servers
    server_descriptions = self._select_servers_loop(
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 218, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: 172.36.0.1:27017: timed out, Timeout: 30s, Topology Description: <TopologyDescription id: 61af3eb03c94e0047bb56ec7, topology_type: Single, servers: [<ServerDescription ('172.36.0.1', 27017) server_type: Unknown, rtt: None, error=NetworkTimeout('172.36.0.1:27017: timed out')>]>
Traceback (most recent call last):
  File "/opt/emba/external/cve-search/./sbin/db_mgmt_json.py", line 22, in <module>
    from lib.Sources_process import CVEDownloads
  File "/opt/emba/external/cve-search/sbin/../lib/Sources_process.py", line 17, in <module>
    from lib.DatabaseLayer import (
  File "/opt/emba/external/cve-search/sbin/../lib/DatabaseLayer.py", line 35, in <module>
    mongo_version = db.command("buildinfo")["versionArray"]
  File "/usr/local/lib/python3.9/dist-packages/pymongo/database.py", line 757, in command
    with self.__client._socket_for_reads(
  File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/mongo_client.py", line 1387, in _socket_for_reads
    server = self._select_server(read_preference, session)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/mongo_client.py", line 1346, in _select_server
    server = topology.select_server(server_selector)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 244, in select_server
    return random.choice(self.select_servers(selector,
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 202, in select_servers
    server_descriptions = self._select_servers_loop(
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 218, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: 172.36.0.1:27017: timed out, Timeout: 30s, Topology Description: <TopologyDescription id: 61af3ed01f494612bdad95e0, topology_type: Single, servers: [<ServerDescription ('172.36.0.1', 27017) server_type: Unknown, rtt: None, error=NetworkTimeout('172.36.0.1:27017: timed out')>]>
Traceback (most recent call last):
  File "/opt/emba/external/cve-search/./sbin/db_updater.py", line 21, in <module>
    from lib.DatabaseSchemaChecker import SchemaChecker
  File "/opt/emba/external/cve-search/sbin/../lib/DatabaseSchemaChecker.py", line 7, in <module>
    from lib.DatabaseHandler import DatabaseHandler
  File "/opt/emba/external/cve-search/sbin/../lib/DatabaseHandler.py", line 4, in <module>
    from lib.DatabasePlugins.config import DatabasePluginLoader
  File "/opt/emba/external/cve-search/sbin/../lib/DatabasePlugins/config.py", line 1, in <module>
    from lib.DatabasePlugins import *
  File "/opt/emba/external/cve-search/sbin/../lib/DatabasePlugins/mongodb.py", line 12, in <module>
    from lib.DatabaseLayer import sanitize
  File "/opt/emba/external/cve-search/sbin/../lib/DatabaseLayer.py", line 35, in <module>
    mongo_version = db.command("buildinfo")["versionArray"]
  File "/usr/local/lib/python3.9/dist-packages/pymongo/database.py", line 757, in command
    with self.__client._socket_for_reads(
  File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/mongo_client.py", line 1387, in _socket_for_reads
    server = self._select_server(read_preference, session)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/mongo_client.py", line 1346, in _select_server
    server = topology.select_server(server_selector)
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 244, in select_server
    return random.choice(self.select_servers(selector,
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 202, in select_servers
    server_descriptions = self._select_servers_loop(
  File "/usr/local/lib/python3.9/dist-packages/pymongo/topology.py", line 218, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: 172.36.0.1:27017: timed out, Timeout: 30s, Topology Description: <TopologyDescription id: 61af3eef37ab0f6687597923, topology_type: Single, servers: [<ServerDescription ('172.36.0.1', 27017) server_type: Unknown, rtt: None, error=NetworkTimeout('172.36.0.1:27017: timed out')>]>

m-1-k-3, 2021-12-07:
Is this a complete fresh installation on a fresh system or was there already an installation on it?
If there was already an installation on this system you need to check the "old" configuration and probably need to adapt it.

Please check the following things:
* external/cve-search/etc/configuration.ini
-> Redis Host configuration should be 127.0.0.1
-> Database Host configuration should be 172.36.0.1

* ifconfig command should show an additional network interface "emba_runs" with ip address 172.36.0.1

m-1-k-3, 2021-12-18:
Closing this for now. If not working we can re-open it

closed 2021-12-18 by m-1-k-3