e-m-b-a/emba #218: pip3 install requirements.txt

torabi12, 2022-02-15:
**Describe the bug**
The emba installer when uses the
pip3 install -r requirements
command this error comes:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
2. See error

**Expected behavior**
pip3 install should be working fine.

**Screenshots**
Added above.

**Desktop (please complete the following information):**
 - OS: [Kali Linux 2022.01, 2021.4 and 2021.3]
 I have tested 3 different Kali versions including the latest one and I got the same issue.

m-1-k-3, 2022-02-15:
which module of the installer is this error from? IF20_cve_search?

m-1-k-3, 2022-02-16:
Looks like broken deps. This should work:
```sudo python3 -m pip install -Iv crackmapexec==5.1.7.dev0```
I will include this into the next update.

torabi12, 2022-02-16:
Thank you the fix command, It was working on Kali 2022.1 with no PROXY.

m-1-k-3, 2022-02-18:
are you running out of disk space?

closed 2022-02-18 by m-1-k-3