e-m-b-a/emba #219: I05-emba_docker_image download with PROXY fails

torabi12, 2022-02-15:
**Describe the bug**
In case of PROXY the default installation of emba stops at I05_emba_docker_image download:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Set PROXY settings
2. EMBA installation (/default mode)
3. See error

**Expected behavior**
The image download should be working fine.

**Screenshots**
Added above.

**Desktop (please complete the following information):**
 - OS: [Kali Linux 2022.01]

m-1-k-3, 2022-02-15:
Currently the proxy installation is not supported. You can disable the STRICT mode from the installer via setting the variable "STRICT_MODE=1" to 0 in installer.sh

Please do a quick check if this helps.

torabi12, 2022-02-15:
Thank you, with STRICT_MODE=0 the installer went through the docker image download.

Best regards,
torabi

m-1-k-3, 2022-02-15:
Thanks for testing and feeback. Good to know for future references and tests.

closed 2022-02-15 by m-1-k-3