e-m-b-a/emba #222: Installation error

FatNinjaRider, 2022-02-19:
**Describe the bug**
Installation failed

**To Reproduce**
Steps to reproduce the behavior:
clone repo
run: sudo ./installer.sh -d

**Expected behavior**
Expected no error

**Screenshots**
ERROR: The Compose file './docker-compose.yml' is invalid because:
networks.emba_runs value Additional properties are not allowed ('name' was unexpected)
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  docker-compose up --no-start
Location:  ./installer/I05_emba_docker_image_dl.sh, line 49
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 49 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 152 -> main -d

**Desktop (please complete the following information):**
 - OS: Lubuntu 20.04

**Additional context**
Add any other context about the problem here.

m-1-k-3, 2022-02-20:
Hi, great that you are trying EMBA on Lubuntu. The error message looks like your docker-compose version does not support the "name" property. What is your installed docker-compose version? You can get it via `docker-compose -v`

FatNinjaRider, 2022-02-20:
my docker-compose is version 1.25.0

Rick

FatNinjaRider, 2022-02-20:
upgraded my docker, emba installed successfully.

Rick

m-1-k-3, 2022-02-20:
Thanks for the information. I will add this detail to the system requirements section in the wiki

closed 2022-02-20 by m-1-k-3