e-m-b-a/emba #228: Installation

keesj-exset, 2022-03-18:
**Describe the bug**

I am getting an error during the installation near  metasploit-framework

**To Reproduce**

1. EMBA installation (default mode)
set -e
git clone https://github.com/e-m-b-a/emba.git
cd emba
sudo ./installer.sh -F

Log;

    default: tcllib will be newly installed.
    default: 
    default: metasploit-framework
    default: ------------------------------------------------------------------------------------------------
    default: Error detected - status code 100
    default: Command:  TOOL_INFO="$(apt show "${1:-}" 2> /dev/null)"
    default: Location:  ./installer/helpers.sh, line 47
    default: Stack Trace:
    default:     [1] print_tool_info(): ./installer/helpers.sh, line 47 -> print_tool_info metasploit-framework 1
    default:     [2] I01_default_apps(): ./installer/I01_default_apps.sh, line 33 -> I01_default_apps
    default:     [3] main(): ./installer.sh, line 156 -> main -F
    default: 
    default: Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues
    default: 
    default: ------------------------------------------------------------------------------------------------
    default: bash: line 5: xit: command not found

"xit" command not found -> exit perhaps?

keesj-exset, 2022-03-18:
I tried this on differnt machines

m-1-k-3, 2022-03-19:
Hi @keesj-exset 

As far as I know the metasploit package is only available on Ubuntu. I recommend you use the docker container with "sudo installer.sh -d". This installs only the cve-search and docker environment on your host. 

If you really prefer using the dev mode you could bypass the errors with disabling strict mode in the installer. But I'm not sure if you are going to get a working installation at the end.

keesj-exset, 2022-03-20:
Hi,

I indeed came to this conclusion. The install script depends on kalilinux. I created a vagrantfile (to generate a virtual machine with emba installed). I did not test much yet but the installation works. I did try a few other approaches documented in the install but quite frankly is was all a bit confusing (if you do not start with a kalilinux base). 

Perhaps .. somebody want to have a look at the vagrant based setup (this would also allow to "throw" away the virtual machine after use."

https://github.com/x7-labs/EMBAbox

m-1-k-3, 2022-03-20:
Cool! I will include a note with a link to your repo in the wiki

m-1-k-3, 2022-03-24:
Now I included a link to EMBAbox in our wiki: https://github.com/e-m-b-a/emba/wiki/Installation#random-installation-topics

Thanks for your great addition to EMBA. Will close this for now

closed 2022-03-24 by m-1-k-3