e-m-b-a/emba #239: Improved storage format?

keesj-exset, 2022-05-10:
**Is your feature request related to a problem? Please describe.**

I am trying to use emba to generate a report . I am combining the results found here with other tools. Initially, the idea was to go through the  CSV files and generate rst or latex report but I am running into small problems. For example

f50_base_aggregator.txt (the text version) is pretty nice in terms of overview but to me it appears the CSV does not contain the same information also while trying to read the CSV it appears that not all the lines in the file have the same amount of fields 

architecture_verified;"ARM"
strcpy_bin;"libicui18n.so";"11"
version_details;kernel;4.9.160;CVEs;788;Exploits;197

The main thing I would like in terms of functionality is to be able to query the data but at this point, I don't know if reading the CSV files is the way to go. 

**Describe the solution you'd like**

GIven there is also already a MongoDB database I was thinking perhaps the csv needs to become json and this can be put in the database. Perhaps just converting to json would be enough . I think it would be great to have a shell/ web front-end where you can query the data (on  file base perhaps?) and get the combined results of different tests. Does Metasploit not do something similar?

**Describe alternatives you've considered**

I am currently just including the .txt files in my report (including colors) and emba is really useful to help determine what to perform for a pentest but it would be nice to be able to also use the findings(filter them) in a report. 
I Have previously written a plugin to include CVE type data into  https://github.com/blacklanternsecurity/writehat but I would currently like to use python to read the data and generate nice graphs/ links etc.

**Additional context**

I hope we can have a nice discussion on the topic. It might also be possible to rewrite some of the shell scripts to store a bit more context( or store the data into a more computer-friendly format) or perhaps I am missing something

m-1-k-3, 2022-05-10:
Hi @keesj-exset, good to hear from you :)
Because of time constraints the CSV reporting is currently not finished. Some of the modules already log to csv, some not. The goal for version 1.1.0 is to get all the modules log the relevant data to csv and use the csv logged data for further processing in other modules. Within this task we also need to re-create the csv report from the f50 module. This was quick shot long time ago to get the data for EMBArk.

m-1-k-3, 2022-05-10:
btw. most of the time it is quite easy to log to csv. I have done a quick test on the F10 modules here: 
https://github.com/e-m-b-a/emba/pull/238/commits/1d2b8069710c7bb44aa6901b3ea62a06bba7308a

m-1-k-3, 2022-05-24:
@keesj-exset Now there are more csv log results included in EMBA. Could you do a check if something you need is missing?

keesj-exset, 2022-05-24:
Will do!

m-1-k-3, 2022-05-25:
Please re-open this issue if something essential is missing

closed 2022-05-25 by m-1-k-3