e-m-b-a/emba #240: Failed install

keesj-exset, 2022-05-12:
**Describe the bug**

Hello, while running the install (in embabox) a fresh install currently fails.

```
    default: [+] IP60_fact_extractor
    default: =================================================================
    default: fact-extractor
    default: Description: Wraps FACT unpack plugins into standalone utility. Should be able to extract most of the common container formats. (EMBA fork)
    default: Download-Size: 34 MB
    default: fact_extractor will be downloaded.
    default: 
    default: FACT-extractor will be downloaded and installed!
    default: Reading package lists...
    default: Building dependency tree...
    default: 
    default: Reading state information...
    default: The following additional packages will be installed:
    default:   libcurl4 libldap-2.5-0
    default: The following NEW packages will be installed:
    default:   libldap-2.5-0
    default: The following packages will be upgraded:
    default:   curl libcurl4
    default: 2 upgraded, 1 newly installed, 0 to remove and 948 not upgraded.
    default: Need to get 869 kB of archives.
    default: After this operation, 581 kB of additional disk space will be used.
    default: Do you want to continue? [Y/n]
    default: Abort.
    default: ------------------------------------------------------------------------------------------------
    default: Error detected - status code 1
    default: Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
    default: Location:  ./installer/IP60_fact_extractor.sh, line 40
    default: Stack Trace:
    default:     [1] IP60_fact_extractor(): ./installer/IP60_fact_extractor.sh, line 40 -> IP60_fact_extractor
    default:     [2] main(): ./installer.sh, line 168 -> main -F
    default: 
    default: Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues
    default: ------------------------------------------------------------------------------------------------

```

It looks like the curl install needs at least a -y flag to install without interaction?
https://github.com/e-m-b-a/emba/blob/master/installer/IP60_fact_extractor.sh#L40

m-1-k-3, 2022-05-12:
Thank you for reporting. Have seen it also currently and going to fix it with the next PR

m-1-k-3, 2022-05-12:
btw. is there a reason for you to use the full install? Why not using the default docker installation?

m-1-k-3, 2022-05-12:
should be fixed.

closed 2022-05-12 by m-1-k-3

keesj-exset, 2022-05-13:
> btw. is there a reason for you to use the full install? Why not using the default docker installation?

Hello, this is just my personal opinion but I feel that If I can not build and modify something I do not have control over it. Docker creates this barrier + I do not trust docker images  (very hard to prove they are built from the sources). 

That said I do not fully understand install script and for example during the install  I get the following message:

```
    default: [+] I05_emba_docker_image_dl                                                                                                                                                                     
    default: =================================================================                                                                                                                                
    default:                                                                                                                                                                                                  
    default: docker.io                                                                                                                                                                                        
    default: Description: Linux container runtime                                                                                                                                                             
    default: Download-Size: 35.1 MB                                                                                                                                                                           
    default: docker.io won't be updated.                                                                                                                                                                      
    default:                                                                                                                                                                                                  
    default: embeddedanalyzer/emba docker image                                                                                                                                                               
    default: Description: EMBA docker images used for firmware analysis.                                                                                                                                      
    default: Download-Size : 4284 MB                                                                                                                                                                          
    default:                                                                                                                                                                                                  
    default: docker.io and the EMBA docker image (if not already on the system) will be downloaded and installed!                                                                                             
    default: Reading package lists...                                                                                                                                                                         
    default: Building dependency tree...     
```

Hence it might even be that while I am installing using the ./installer.sh -F method I might actually be using the docker image after all

keesj-exset, 2022-05-13:
> should be fixed.

I can confirm the build now finishes