e-m-b-a/emba #247: Docker-Compose  ('name' was unexpected) fix

erd0spy, 2022-05-17:
I got docker-compose name error on updated Ubuntu 20.04 while installing.

According to the solution I found on stackoverflow, adding this to the install script will fix the problem:

```bash
sudo curl -L https://github.com/docker/compose/releases/download/1.28.5/docker-compose-`uname -s`-`uname -m` -o /usr/local/bin/docker-compose
sudo chmod +x /usr/local/bin/docker-compose
```

**Referance:**
https://stackoverflow.com/questions/58155523/unable-to-give-network-name-in-docker-compose

m-1-k-3, 2022-05-18:
Have you got the following warning during installation?

[image]

Probably we should print more details here. But this check indicates that your docker-compose environment is outdated. I will improve this message

m-1-k-3, 2022-05-18:
linked this issue from the wiki: https://github.com/e-m-b-a/emba/wiki/Installation#random-installation-topics

p4cx, 2022-05-19:
I checked the docker-compose version in the package archives and it's 1.25.0: https://packages.ubuntu.com/focal/docker-compose

EMBA uses some features of docker-compose, which were introduced in a more modern version and at the moment you have to use docker-compose 1.28.5 or higher to be able to use EMBA.

To get the best experience with EMBA, please use a modern Kali Linux. Alternatively, you can install a higher version of docker-compose manually and try it again. But if you want to try the second option, we can't support you. Have fun with EMBA 👍

closed 2022-05-19 by p4cx

m-1-k-3, 2022-05-19:
I did a check with the mentioned workaround from the initial post and EMBA works fine on the Ubuntu 20.04-LTS. The improved note during installation is already merged.