e-m-b-a/emba #256: embeddedanalyzer/emba docker image ERROR INSTALLING

rodry7, 2022-06-16:
**Describe the bug**
Whenever I try to install the embeddedanalyzer/emba docker image it just gives me the error: failed to register layer: Error processing tar file(exit status 1): write /external/arachni/arachni-1.6.1.3-0.6.1.1/.system/usr/lib/ruby/2.7.0/rdoc/generator/template/darkfish/fonts/SourceCodePro-Regular.ttf: no space left on device
------------------------------------------------------------------------------------------------
Error detected - status code 1

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
2. sudo ./installer.sh -d 
4. See error

**Expected behavior**
It should install without any problem I don't know why am I getting this error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Kali Linux 2022

p4cx, 2022-06-17:
How much space do you have on your machine? Your error message tells us that there is "no space left on device". 
You can check it with `df` on Kali Linux.

We are suggesting to have at least over 25 GB of empty space on your machine, because the docker container alone should take about 15 GB and it is never a bad idea to have some reserves.

rodry7, 2022-06-17:
I think I had 80GB on the kali VM
[image]
[image]

p4cx, 2022-06-18:
Somethings wrong with your disk in your VM. If I divide 18964304/1024/1024, then it gives about 18GB (or just use `df -h`). I have no idea if this is the right way to calculate that, but I checked it on my machine, it comes quite near to the output of `df -h`, therefore we can say, that the error message is not misleading and you haven't enough space on your disk. Downloading and extracting the Container needs at least more than 18GB space. I hope you can resolve your disk size issues.

m-1-k-3, 2022-06-19:
As far as I can see in the output of the df command you only have 18GB of disk space and around 3GB are free. This is not enough for EMBA. Please check the prerequisites here: https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

closed 2022-06-19 by m-1-k-3