e-m-b-a/emba #260: Installer fails due to libssl issues when installing mongodb-org-* stuff

HanEmile, 2022-06-30:
**Describe the bug**
I wanted to try out emba after the talk at troopers, but mongodb org stuff can't be built due to some libssl issued (see the output of the installer at the very bottom of this issue)

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
    I cloned the repo and executed `sudo ./installer.sh -d` (It's a yolo vm)
2. ~~Use the firmware available here: <Insert URL>~~
4. ~~Start EMBA with the following parameters: sudo ./emba.sh <insert further parameters>~~
5. 
6. ~~See error~~

**Expected behavior**
The installer should run through without problems

**Screenshots**
(output of installer at the very bottom)

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 host

**Additional context**

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mongodb-org-mongos : Depends: libssl1.1 (>= 1.1.0) but it is not installable
 mongodb-org-server : Depends: libssl1.1 (>= 1.1.0) but it is not installable
 mongodb-org-shell : Depends: libssl1.1 (>= 1.1.0) but it is not installable
E: Unable to correct problems, you have held broken packages.
------------------------------------------------------------------------------------------------
Error detected - status code 100
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer/IF20_cve_search.sh, line 112
Stack Trace:
    [1] IF20_cve_search(): ./installer/IF20_cve_search.sh, line 112 -> IF20_cve_search
    [2] main(): ./installer.sh, line 191 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

------------------------------------------------------------------------------------------------
```

m-1-k-3, 2022-06-30:
Currently we only support Kali Linux as host (see the prereq here https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites). 

But, we are working on supporting Ubuntu also ... And I think your issue is already addressed here https://github.com/e-m-b-a/emba/pull/258

HanEmile, 2022-06-30:
maybe then replace the "strongly recommend" phrase in the prerequisites wiki page by "require".

m-1-k-3, 2022-06-30:
Indeed. Will do this.
In the meantime you could give the PR a try if this solves your issue

m-1-k-3, 2022-06-30:
The documentation is now fixed. I will do some tests of the PR the next days.

m-1-k-3, 2022-07-15:
From now on EMBA should install quite fine on Ubuntu ubuntu:jammy (22.04 LTS)

closed 2022-07-15 by m-1-k-3