e-m-b-a/emba #270: Install error (on new kali vm)

peepuk, 2022-07-16:
I get an error on a fresh kali vm, the screenshot gives details of the error..

**systeminfo:**
  └─$ lsb_release -a
    No LSB modules are available.
    Distributor ID: Kali
    Description:    Kali GNU/Linux Rolling
    Release:        2022.2
    Codename:       kali-rolling

   *80Gb disk, 16 processors, 8Gb mem

**reproduce with**

git clone https://github.com/e-m-b-a/emba.git
cd emba
sudo ./installer.sh -d

*E: Failed to fetch http://ftp2.nluug.nl/os/Linux/distr/kali/dists/kali-rolling/contrib/Contents-amd64.gz  File has unexpected size (157540 != 157526). Mirror sync in progress? [IP: 145.220.21.40 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.*

**Screenshot**

[image]

p4cx, 2022-07-16:
That is a strange issue. It seems it happens while running `apt-get -y update`.  What happens if you run this command manually? Same error?

m-1-k-3, 2022-07-16:
Could you please wait an hour or so and try it again? See https://superuser.com/questions/1485595/q-how-to-fix-apt-get-update-error-mirror-sync-in-progress

peepuk, 2022-07-16:
Yes, thank you both, the installation is continuing.

I will close this issue.

closed 2022-07-16 by peepuk