e-m-b-a/emba #284: dbus-launch error

torabi12, 2022-08-02:
**Describe the bug**
There is an error when I start EMBArk:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. EMBArk installation (default mode)
2. Start EMBArk with the following parameters: sudo ./run-server.sh
3. See error

**Expected behavior**
Child process should be available.

**Desktop (please complete the following information):**
 - OS: [Ubuntu 22.04 LTS Desktop]

m-1-k-3, 2022-08-02:
Is this EMBA or EMBArk related? Can you trigger this error also in EMBA or only in an EMBArk test?

torabi12, 2022-08-02:
I am testing EMBArk installation and I reported to Benedikt this error and he suggested to open a bug for EMBA to track it.

torabi12, 2022-08-03:
Hello All,

I installed only EMBA and I did a short test today and I got similar errors like above:

[image]

[image]

Ubuntu Desktop 22.04 LTS
git clone https://github.com/e-m-b-a/emba.git

Regards,
Torabi

m-1-k-3, 2022-08-03:
Not sure if this is the new notification system. Could you try setting the following option to 1: https://github.com/e-m-b-a/emba/blob/a2986346274258588a735bd8e4673cb63a9217a5/emba.sh#L310

torabi12, 2022-08-03:
I changed the 0 to 1 and the error didn't come.

m-1-k-3, 2022-08-03:
Ok, looks like some Ubuntu special issue. Are the following commands available/installed? 
* inotifywait
* notify-send

Are you working in a graphical environment?

torabi12, 2022-08-03:
Both are available on Ubuntu 22.04 LTS. I am using GUI and default installation.

m-1-k-3, 2022-08-03:
ok, thanks for reporting. I will do some further testing.

m-1-k-3, 2022-08-03:
Could you check if the following packages are installed:
* dbus
* dbus-x11

Should work with a command like the following:
`dpkg -l | grep dbus`
If these are not installed please install it and try again.

m-1-k-3, 2022-08-03:
ok, I will setup a Ubuntu and take a look. Thanks for testing

m-1-k-3, 2022-08-04:
just as a quick note ... if you are usually using the default profile, you can use the default profile without notifications enabled (https://github.com/e-m-b-a/emba/blob/master/scan-profiles/default-scan-no-notify.emba)

jgamblin, 2022-08-05:
Just as an FYI, I am getting the same error using EMBA.

[image]

m-1-k-3, 2022-08-05:
Thank you @jgamblin for reporting this. As a quick workaround you can use the profile "./scan-profiles/default-scan-no-notify.emba" in the following way:

```
└─$ sudo ./emba.sh -f ~/firmware.bin -l ~/emba_logs -p ./scan-profiles/default-scan-no-notify.emba 
```

You can also disable this feature by setting the following option to 1:
https://github.com/e-m-b-a/emba/blob/76cdefad70d805aefa40c6f0552461c85e4bb7a6/emba.sh#L310

I will do some further tests soon and I am sure we can find and fix the issue.

m-1-k-3, 2022-08-05:
Please try installing the package "notification-daemon":

```
sudo apt-get install notification-daemon
```
If it is then working you are fine. If not you need to create the following file:

```
/usr/share/dbus-1/services/org.freedesktop.Notifications.service
```
and add the following content:

```
[D-BUS Service]
Name=org.freedesktop.Notifications
Exec=/usr/lib/notification-daemon/notification-daemon
```

Please respond with your exact steps that helped in this thread. Thank you

torabi12, 2022-08-05:
1st I just installed the notification-daemon:

[image]

it did not solve the issue:
[image]

the I created the file with the content, but the error appeared again in the terminal more times:
[image]

This is the file I added:

[image]

jgamblin, 2022-08-05:
I am seeing the same behavior.

m-1-k-3, 2022-08-05:
hmmm ... checked my history and found the following stuff I did:

```
apt-get install dbus dbus-x11
apt-get install libnotify-cil-dev
apt-get install notification-daemon
```
And then modified the mentioned file. Was it the package `libnotify-cil-dev` that I missed? Could you check this?

torabi12, 2022-08-05:
I installed dbus-x11 and libnotify-cil-dev also but I got the error again:
[image]

I had the same output when libnotify package was not installed but I tried with it also.

I usualy set back to a clean snapshot the VM to have a perfect test environment. The error only disappeared when I set the 0 to 1 in emba.sh in line 310, or somewhere there:)

m-1-k-3, 2022-08-05:
Probably we should set it default to 0. On the other hand we need to solve this issue. Will give it a next try tomorrow

m-1-k-3, 2022-08-06:
Please check #286

p4cx, 2022-08-07:
This issue should now be resolved and notifications should work in Ubuntu. Just run `installer.sh -d` and then it should be good to go. If there are other problems in this direction, please reopen this issue or create a new one.

torabi12, 2022-08-15:
OS: Ubuntu 22.04.1 LTS
emba installation was fine I didn't have any errors during the process.

I executed a short scan:
$ sudo ./emba.sh -f ../WRT54GS_3.17.4_US_code.bin -l /home/arpadb/WRT54GS -p ./scan-profiles/full-scan.emba

I got this error or similar error messages many times:
[image]

[image]

[image]

[image]

There were some other strange lines in the output where emba displayed a usage of the actual tool:
[image]

Please execute a similar scan for an older Linksys file if you would like to verify the above messages.
https://cyberforat.squat.net/openwrt/binary_images/linksys_official/WRT54GS_3.17.4_US_code.zip

m-1-k-3, 2022-08-17:
The openssl error is occuring during the user-mode emulation. This is an expected message as the emulator enumerates through different parameters like -h, -v, -V, --help and so on.

m-1-k-3, 2022-08-17:
The other issue looks like a known Ubuntu issue: https://www.google.com/search?q=gdbus+notifications+exceeded+maximum+number+of+notifications&rlz=1C1GCEB_enDE995DE995&oq=gdbus+notifications+exceeded+maximum+number+of+notifications&aqs=chrome..69i57.15575j0j7&sourceid=chrome&ie=UTF-8

torabi12, 2022-08-23:
Thank you, for me it is OK, I just wanted to share with you to see if its emba related or not.

closed 2022-08-23 by m-1-k-3