e-m-b-a/emba #287: Scan Wont Finish On Ubuntu

jgamblin, 2022-08-05:
**Describe the bug**
Attempting to run a scan I am stuck on S115 for last 4+ hours with a alternating screen of `CVE-Search - Ok` and `CVE-Search - Testing`

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation: **Default**
2. Use the firmware available here: https://static.tp-link.com/2018/201803/20180327/TL-WR902AC(US)_V3_171229.zip
3. Start EMBA with the following parameters: sudo ./emba.sh -l ./log -f ./firmware/TL-WR902AC(US)_V3_171229.zip -W
5. See error

**Expected behavior**
The scan to finish. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 
 - [Have been helping try to troubleshoot issue 284](https://github.com/e-m-b-a/emba/issues/284)

torabi12, 2022-08-05:
Hello Jerry,

May I ask you the HW resources of your test machine?
I suppose that is a VM running in Virtual box or VMware, but what about the size of vRAM? vCPUs?

I have executed  alot of scans with EMBA or with EMBArk and I use 26GB RAM and 8 CPUs for the VM. The scan can be very very long, sometimes it can reach up to 2 days.

m-1-k-3, 2022-08-05:
I will run a testscan of the firmware ... we will see

jgamblin, 2022-08-05:
@torabi12 8 Cores and 16480MB of Memory.  How long a scan would take would be good information for the WiKi.  I was not expecting it to take days, but it makes sense. 

[image]

torabi12, 2022-08-05:
I also tried to scan this FW, I extracted the ZIP and I only gave the BIN file to EMBA but it did nto start:
[image]

I used this command for scan:
$ sudo ./emba.sh -f ../TL-WR902ACv3.bin -f /home/arpadb/TL-WR902ACv3_2 -W

I don't know what should be not OK now:(

m-1-k-3, 2022-08-05:
My VM with 20 Gigs of RAM and 8 Cores (currently testing on my Kali) is now running less than an hour and already finished s115 (the emulator) and s116 (the version detection of the emulation results). I have started it with the default profile:

```
sudo ./emba.sh -f /home/m1k3/firmware_stuff/Firmware_images/TL-WR902AC(US)_V3_171229.zip -l /home/m1k3/firmware_stuff/emba_logs_tl-wr902ac -S -p ./scan-profiles/default-scan-emulation.emba
```
I will keep you updated ...

torabi12, 2022-08-05:
binwalk is not installed by default on Ubuntu 22.04 LTS. After installing it I was able to start the scan for the bin file also.

m-1-k-3, 2022-08-05:
you do not need to install binwalk on your host. All these deps should be installed in the docker container.

m-1-k-3, 2022-08-05:
@jgamblin your test is running without threading. This will take a while. Could you try restarting your test with `-t` option or (in my opinion the best solution) start it with `-p scan-profiles/default-scan.emba`. If this works find you could also try to run the profile `scan-profiles/default-scan-emulation.emba`

The default profile sets some further options like: `-t -s -z -E -W`
The default profile with emulation enabled sets additionally the option `-Q`

jgamblin, 2022-08-05:
I started it with the profile, and it just started to scroll this: 
[image]

I am running it with just `-t` to see if I can get one complete run.

m-1-k-3, 2022-08-05:
hell yea ... it works :) same here! This is the user-mode emulator (s115 module):

[image]

Give it a bit of time. In my environment it is nearly finished:
```
[!] Pre-checking phase ended on Fri Aug  5 20:24:51 CEST 2022 and took about 00:00:57 
[!] Testing phase ended on Fri Aug  5 21:12:41 CEST 2022 and took about 00:48:47 
[!] System emulation phase ended on Fri Aug  5 22:49:03 CEST 2022 and took about 02:25:09 
```

[image]

torabi12, 2022-08-05:
Good evening / afternon,

My test ended just now with these results:
[image]

This was my init command:
$ sudo ./emba.sh -f ../TL-WR902ACv3_US_0.9.1_0.2.bin -l /home/arpadb/TL-WR902ACv3 -S -p ./scan-profiles/default-scan-emulation.emba 

I don't like to feed emba with zip and other files, I just add the main FW file to have clean scan.

Regards,
torabi

m-1-k-3, 2022-08-06:
Looks good ... Test ended. Then there is the error from the notification environment on Ubuntu. Go to the log directory and there should be all the results.

m-1-k-3, 2022-08-06:
Please check #286

jgamblin, 2022-08-06:
Thank You!  I was finally able to complete a firmware scan this afternoon. 

[image]

m-1-k-3, 2022-08-08:
I think it is solved for now. I will note that we need to improve our docs. New users should find the right way easier.

Thanks for pointing me on this issue. Will close this issue for now.

closed 2022-08-08 by m-1-k-3