e-m-b-a/emba #289: Kali Linux 2022.3 released

m-1-k-3, 2022-08-18:
**Is your feature request related to a problem? Please describe.**

Kali Linux 2022.3 released here: https://www.kali.org/blog/kali-linux-2022-2-release/
We need to test EMBA on it

**Testcases:**

- [ ] Default/docker installation working
  - [ ] EMBA run with profile default-scan in strict mode (-S)
- [ ] Full installation working
  - [ ] EMBA run in dev mode with options -s -z -S -D -E -t -W
  - [ ] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W
- [ ] Docker build
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)

torabi12, 2022-08-20:
I did a default installation on Kali 2022.3 with emba and I didn't get any errors. I also checked this scan:
$ sudo ./emba.sh -f ../WRT54GS_3.17.4_US_code.bin -l /home/kali/WRT54GS -p ./scan-profiles/full-scan.emba

and I got the final report without problems:
[image]

m-1-k-3, 2022-08-20:
The current cwe-checker installation is broken. I will open a dedicated issue for this

m-1-k-3, 2022-08-23:
EMBA is running quite smooth on the newest Kali Linux v2022.03. The cwe-checker installation will be handled in a separate issue. An updated docker container will be created and published soon

closed 2022-08-23 by m-1-k-3