e-m-b-a/emba #295: installer.sh, problem with jq

nuschpl, 2022-08-31:
=================================================================

To use EMBA, some applications must be installed and some data (database for CVS for example) downloaded and parsed.

These applications will be installed/updated:

jq
Description: lightweight and flexible command-line JSON processor
Download-Size: 52,5 kB
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer/helpers.sh, line 64
Stack Trace:
    [1] print_tool_info(): ./installer/helpers.sh, line 64 -> print_tool_info jq 1
    [2] I01_default_apps_host(): ./installer/I01_default_apps_host.sh, line 24 -> I01_default_apps_host
    [3] main(): ./installer.sh, line 182 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

nuschpl, 2022-08-31:
Besides above error I have jq installed and in $PATH for current shell

nuschpl, 2022-08-31:
It probably caused by the tool scrapping command line output and not assuming shell command output won't be localized

nuschpl, 2022-08-31:
It worked after adding above, at least for my case. More command around seems to depend on it, but the pull request contains only the one change I've tested

m-1-k-3, 2022-09-01:
Thank you for your Feedback. Indeed we never tested EMBA with other language settings. What are your current settings and what is the system you are running EMBA in?

nuschpl, 2022-09-01:
I'm on Ubuntu 22.04, with Polish locale, precisely:
`$ locale
LANG=pl_PL.UTF-8
LANGUAGE=
LC_CTYPE="pl_PL.UTF-8"
LC_NUMERIC="pl_PL.UTF-8"
LC_TIME="pl_PL.UTF-8"
LC_COLLATE="pl_PL.UTF-8"
LC_MONETARY="pl_PL.UTF-8"
LC_MESSAGES="pl_PL.UTF-8"
LC_PAPER="pl_PL.UTF-8"
LC_NAME="pl_PL.UTF-8"
LC_ADDRESS="pl_PL.UTF-8"
LC_TELEPHONE="pl_PL.UTF-8"
LC_MEASUREMENT="pl_PL.UTF-8"
LC_IDENTIFICATION="pl_PL.UTF-8"
LC_ALL=`
Maybe better idea would be to not depend on command line output at all as it may change with time. E.g python apt api seems pretty straightforward:
https://stackoverflow.com/questions/17537390/how-to-install-a-package-using-the-python-apt-api/17538002#17538002

m-1-k-3, 2022-09-01:
Thank you for this suggestion. I will note this on our low prio todo list.
For quick fixing your PR will be quite fine.

m-1-k-3, 2022-09-01:
Thank you for your fix. Was verified in default installation.

closed 2022-09-01 by m-1-k-3