e-m-b-a/emba #297: Many links on html-report index.html contain original filename extensions causing 404 errors

n0x08, 2022-09-01:
**Describe the bug**
The index.html page of web reports has many instances where the page links are things like:

file.txt.html
file.csv.html
file.html.html

These cause 404 when clicked & require manually removing the additional file extension from the URL.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (using version updated 8/31/2022)
2. Run EMBA with -W flag
3. Copy html-report directory structure to web server root folder
4. Attempt to browse links on index.html page
5. Receive 404 errors for many pages

**Expected behavior**
Links on index.html page should accurately reflect sub-pages in the report directory

**Screenshots**
[image]

**Desktop (please complete the following information):**
Linux emba 5.15.0-46-generic #49-Ubuntu SMP Thu Aug 4 18:03:25 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

**Additional context**
Add any other context about the problem here.

dgutson, 2022-09-01:
Tag @qequ

p4cx, 2022-09-01:
Thank you for reporting this issue. I'll take a look at that soon!

m-1-k-3, 2022-09-02:
@dgutson I did a grep on index.html but I have not found the orig. filename with the .txt in it:

[image]

Could you provide further details how to get these results?

dgutson, 2022-09-02:
@m-1-k-3 actually the OP was @n0x08  :) That goes to him.

n0x08, 2022-09-02:
I ran emba with this command: sudo ./emba.sh -W -l ./firmware.log -f ~/firmware.dav -p scan-profiles/default-scan-emulation.emba

And when I check the index.html I see the .txt.html and .html.html links

[image]
[image]

m-1-k-3, 2022-09-02:
Ha ... got it! Something with your command line is causing this issue.

As a quick work a round could you please try to call EMBA the following way:
```
sudo ./emba.sh -f ~/firmware.dav -l ~/firmware_log -p ./scan-profiles/default-scan.emba 
```
* The -W parameter is already set in the profile
* firmware.log -> please remove the "."
* Please store the logs outside of the emba directory

I need to verify what the actual issue is but with this command it should work and produce a correct report

n0x08, 2022-09-02:
I still get the same behavior; I used this start command: nate@emba:~/emba$ sudo ./emba.sh -f ~firmware.dav -l ~/firmware.log -p scan-profiles/default-scan.emba

[+] EMBA start command: ./emba.sh -f /home/nate/firmware.dav -l /home/nate/firmware.log -p scan-profiles/default-scan.emba

I tested the same firmware image and the number of corrupted links changed; this is the count of the test run with your suggested syntax:

[image]

And this is the count from my initial report:
[image]

n0x08, 2022-09-02:
I get the same behavior when I omit .log from the logging directory. Same statistics on corrupted links as in my first screenshot above.

m-1-k-3, 2022-09-02:
Are there any old logfiles in the emba directory e.g. from your initial tests? e.g., /home/nate/emba/firmware_log or /home/nate/emba/firmware.log

Could you please clean the emba dir (e.g., move them to a different place) and give it another try?
btw. you can also use the dots in your log directory and the double -W parameter also does not matter. Just remove it from the emba directory. If you can confirm this behavior I will improve our check on this case.

dgutson, 2022-09-02:
@m-1-k-3 maybe it'll be easier if we try to solve this ourselves given that we have the input data to reproduce the issue.
@qequ could you please try to reproduce Nate's issue? If so, I propose to assign this to @qequ and let him fix it.

n0x08, 2022-09-02:
I cleared out *all* log directories from the emba folder and the parent and now it doesn't corrupt the URLs. I had an unrelated log directory from a different firmware in my emba folder that I had forgot about. I was removing the log directories before each iteration of this testing so I guess it was the old one causing the issue. Thank you for the help!

dgutson, 2022-09-02:
Just curious, question to maintainers: what's the relationship between logs/logs dir, and URLs corruption?

m-1-k-3, 2022-09-03:
the question for next week ;)

p4cx, 2022-09-04:
I figured it out. EMBA checks in the web report whether the modules really exist and found the old log files instead of the modules themselves and wrongly linked them. #300 should solve the problem.

closed 2022-09-04 by p4cx