e-m-b-a/emba #298: Excessive permissions

nuschpl, 2022-09-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. default-mde
3. Use the firmware available here: N/A
4. Start EMBA with the following parameters: ./emba.sh -f anyfirmware.bin
6. See error
With docker enabled this script needs root privileges

**Expected behavior**
There is no reason software of this kind should need root privileges every time. Especially when most of work is done within containers

m-1-k-3, 2022-09-01:
Currently EMBA needs root privs for multiple extraction modules, as well as for the chroot environment from the user-mode emulator.

nuschpl, 2022-09-01:
What kind of extraction needs root ? If you need chroot there is no problem with launching chroot command itself with sudo, instead of running full framework full of uncertain tools as root.
Why won't you preciselty state those issues so other could contribute to those subissues ? You are building security tool and ignoring basic security principles - AFAIR emulator is optional and the message says it's docker related . What is purpose for running containerized environmen at all if we don't care about security

m-1-k-3, 2022-09-01:
> Why won't you preciselty state those issues so other could contribute to those subissues ?

Keep calm guy ... if you ask you get the info.

> You are building security tool and ignoring basic security principles 

On the other side you do not need to use it at all.

For all that are interested the following modules currently need root privs:
* P10
* P14
* P18
* P19
* P20
* S115
* L10

Probably other modules also need root privs.

farmmanic, 2024-03-16:
for sure its recommended to run nested vm's 
i run the root commands in the docker image on the bottom of those nested vm's

nuschpl, 2024-04-14:
How the #wontfix and #Research tags come all together, does it mean there is an intent to change current behaviour (ongoing work or PR accept) or not ?

m-1-k-3, 2024-04-14:
#research -> further research work needed to identify all the affected areas and start repairing them or find some solution
#wontfix -> we currently do not have the resources to fix this within the core team

If you would like to start digging into this issue you are welcome. We can use this issue for further discussions, as well as a dedicated disussion in the gh discussion area.

m-1-k-3, 2025-05-16:
I am not very familiar with LXC containers. So, give it a try and if we can support you just ping us.