e-m-b-a/emba #299: Incorrect handling of paths - Windows style on Linux

nuschpl, 2022-09-01:
**Describe the bug**
Somewhere in the middle of binwalk extraction:

> 
> DECIMAL       HEXADECIMAL     DESCRIPTION
> --------------------------------------------------------------------------------
> 0             0x0             Zlib compressed data, best compression
> 1268          0x4F4           PNG image, 1600 x 554, 8-bit/color RGB, non-interlaced
> 2258          0x8D2           Zlib compressed data, best compression
> 260244        0x3F894         JPEG image data, JFIF standard 1.02
> 382164        0x5D4D4         HTML document header
> 396770        0x60DE2         Copyright string: "Copyright<span"
> 397111        0x60F37         Copyright string: "copyright<span style='letter-spacing:-.25pt'>"
> 415373        0x6568D         HTML document footer
> 415412        0x656B4         HTML document header
> 416384        0x65A80         HTML document footer
> 
> [... CUT ... ] 
> md5sum: '/log/firmware/firmware_binwalk_emba/_firmware.extracted/SUM\\company\\css\\company.css': No such file or directory
> md5sum: '/log/firmware/firmware_binwalk_emba/_firmware.extracted/IP\\web\\assets\\oem\\default\\oem.css': No such file or directory
>[.. CUT ...]
> 
> 

The root cause if that some multilevel path files are created with '\' instead of '/' on Linux (Windows style instead of UNIX style) so instead of multidirectory structure expected by emba I'm getting long full path filename with '\' encoded as part of it:
`/IoT Security/emba/logs/firmware/firmware_binwalk_emba$ ls -l
razem 36
drwxr-xr-x 3 root root 4096 wrz  1 15:29  _1A2C4E.extracted
drwxr-xr-x 3 root root 4096 wrz  1 15:29  _3DE.extracted
drwxr-xr-x 2 root root 4096 wrz  1 15:29  _firmware.extracted
drwxr-xr-x 2 root root 4096 wrz  1 15:29 '_IP\web\assets\oem\default\images\vendor_log_left_wht.png.extracted'
drwxr-xr-x 2 root root 4096 wrz  1 15:29 '_IP\web\assets\oem\default\images\IP_Login.png.extracted'

`
Then emba fails in md5sum command. 
**To Reproduce**
Extract some UEFI firmware, observe when binwalk is launched under the hood

**Expected behavior**
Regardless if this is binwalk or emba issues, paths should be normalised

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04

**Additional context**
Add any other context about the problem here.

m-1-k-3, 2022-09-01:
Can you provide the used firmware for verification?

m-1-k-3, 2022-09-15:
Closing this issue as no firmware for testing was provided. Please re-open with a firmware for reproducing this issue.

Thank you.

closed 2022-09-15 by m-1-k-3