e-m-b-a/emba #301: Reinstall of emba

torabi12, 2022-09-08:
**Describe the bug**
I would like to ask you how to reinstall the latest emba on Ubuntu OS? EMBArk has an option -r, which can be used, but I don't know how to do this if an older emba has been installed.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.1 Desktop

m-1-k-3, 2022-09-08:
You are asking for the installed EMBA version within EMBArk? Why would you reinstall EMBA in there?
The EMBA version which is installed by the EMBArk installer is usually tested with EMBArk. With newer versions you could run into issues.

torabi12, 2022-09-08:
I have only emba installed this system, and there are other firmware scannings apps as well, but no EMBArk.
I have emba, firmwalker, firmware-mod-kit and others on the same system and emba was working fine when I installed it, but a few days later the cve-search is not working, I get this error when I want to update emba:
[image]

If I start a scan in command line I get error message about mongodb is not working and emba is not able to start it and I don't know which other application made the problem.

I just want to reinstall full emba, maybe it makes it good again, because I do not want to reinstall the entire system.

m-1-k-3, 2022-09-08:
First of all there is no automatic reinstall procedure for EMBA. But, this is a good idea and we should make this happen.

In your environment it looks like there is an issue with your cve installation or at least with redis and mongod. Let's debug it a bit:

Please check your disk space with `df -h`
Your system should have at least 4gig of RAM to work properly with redis.

Then try to start redis manually:

```
└─$ sudo /etc/init.d/redis-server status                          
[sudo] Passwort für m1k3: 
○ redis-server.service - Advanced key-value store
     Loaded: loaded (/lib/systemd/system/redis-server.service; disabled; preset: disabled)
     Active: inactive (dead)
       Docs: http://redis.io/documentation,
             man:redis-server(1)
                                                                                                                                                  
┌──(m1k3㉿kali)-[~/git-repos/emba_forked]
└─$ sudo /etc/init.d/redis-server start 
Starting redis-server (via systemctl): redis-server.service.
                                                                                                                                                  
┌──(m1k3㉿kali)-[~/git-repos/emba_forked]
└─$ sudo /etc/init.d/redis-server status
● redis-server.service - Advanced key-value store
     Loaded: loaded (/lib/systemd/system/redis-server.service; disabled; preset: disabled)
     Active: active (running) since Thu 2022-09-08 17:11:15 CEST; 2s ago
       Docs: http://redis.io/documentation,
             man:redis-server(1)
   Main PID: 89239 (redis-server)
     Status: "Ready to accept connections"
      Tasks: 5 (limit: 23520)
     Memory: 11.8M
        CPU: 84ms
     CGroup: /system.slice/redis-server.service
             └─89239 "/usr/bin/redis-server 127.0.0.1:6379"

Sep 08 17:11:15 kali systemd[1]: Starting Advanced key-value store...
Sep 08 17:11:15 kali systemd[1]: Started Advanced key-value store.
```

Next, mongod:

```
└─$ sudo service mongod status          
└─$ sudo service mongod start 
```
If you are running into issues with mongod you can check the logs while restarting it
```
sudo tail -f /var/log/mongodb/mongod.log
```

Afterwards you should have a similar netstat output (netstat -anpt):

```
tcp        0      0 172.36.0.1:27017        0.0.0.0:*               LISTEN      -                   
tcp        0      0 127.0.0.1:6379          0.0.0.0:*               LISTEN      -                   
tcp6       0      0 ::1:6379                :::*                    LISTEN      -     
```

If this is the case you should be able to query your cve database from your EMBA directory with the following command
```
└─$ ./external/cve-search/bin/search.py -p busybox      
```

Please let me know where everything is failing ;)

torabi12, 2022-09-08:
Thank you very much!

The HW resources are fine:
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           2,6G   11M  2,6G   1% /run
/dev/sda3       196G   57G  130G  31% /
tmpfs            13G     0   13G   0% /dev/shm
tmpfs           5,0M     0  5,0M   0% /run/lock
tmpfs            13G     0   13G   0% /run/qemu
/dev/sda2       512M  5,3M  507M   2% /boot/efi
tmpfs           2,6G  4,7M  2,6G   1% /run/user/1000
vmhgfs-fuse     3,7T  3,7T   31G 100% /mnt/hgfs

$ free -h
               total        used        free      shared  buff/cache   available
Mem:            25Gi       1,3Gi       4,7Gi        63Mi        19Gi        23Gi
Swap:          2,0Gi          0B       2,0Gi

but I have these:

[image]

I can not start redis and mongodb :(

It used to work yesterday.

m-1-k-3, 2022-09-08:
is something already listening on the ports?
```
netstat -anpt
```
Could you please also monitor the logs during restarting of the services (in a different console):
```
sudo tail -f /var/log/mongodb/mongod.log
sudo tail -f /var/log/redis/redis-server.log
```

m-1-k-3, 2022-09-08:
btw. I have adopted the title of this issue
Could you please open another issue with the initial idea of a reinstallation option in the installer?

torabi12, 2022-09-08:
Netstat gave this:
[image]

Bad news is that both log files are missing:
[image]

Thank you the title update!

torabi12, 2022-09-08:
Excuse me I am not sure emba was working yesterday, because I spent the whole day with the installation of the entire system and I didn't start a test with emba, I just checked the installation logs more or less and I went to the next FW app installation.

m-1-k-3, 2022-09-08:
ok, probably something of the installed tools has killed your installation of mongod and redis. I would recommend doing a reboot and check it again. If nothing changes you could try create the directories manually. If this also don't help remove and purge the mongod and redis packages (you can find them for example with `dpkg -l | grep mongo`) and follow the installation instructions at cve-search to try to recover it: https://cve-search.github.io/cve-search/getting_started/installation.html

m-1-k-3, 2022-09-08:
here are the permissions for manual recreation:

```
drwxr-s---  2 redis             adm                 4096 10. Jul 00:00 redis
drwxr-xr-x  2 mongodb           mongodb             4096  6. Jul 14:29 mongodb
```

torabi12, 2022-09-08:
This part is strange:
[image]

I have this and apt update is not working:(

[image]

torabi12, 2022-09-09:
I was wondering what I did after emba installation and I found omnly one interesting event.

I would like to install
https://github.com/KeenSecurityLab/BinAbsInspector

it requires Ghidra 10.1.2 and Java 11 JDK and z3 4.8.15
https://github.com/Z3Prover/z3

The prebuild z3 package is not OK on Ubuntu 22.04.1 because the g++ version is different I have to build an actual one, I used these commands:

https://github.com/Z3Prover/z3/archive/refs/tags/z3-4.8.15.tar.gz
$ tar xvf z3-4.8.15.tar.gz
$ cd z3-z3-4.8.15/
$ python scripts/mk_make.py --java
$ cd build
$ make
$ sudo make install
$ cd z3-z3-4.8.15/build
$ sudo cp *.so /usr/local/lib/ && sudo cp *.so /lib

Maybe this caused something on the VM.

m-1-k-3, 2022-09-09:
In your screenshot you have written everything in one line without `&&`

You probably need something like the following separated commands:
```
wget --no-check-certificate -qO - https://www.mongodb.org/static/pgp/server-4.4.asc | sudo apt-key add -
echo "deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 multiverse" | sudo tee /etc/apt/sources.list.d/mongodb-org-4.4.list
apt-get update -y
```

m-1-k-3, 2022-09-09:
O now I got your point ... please remove your impish entry completely and start over with the following:

```
        wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb -O external/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
        wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb -O external/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
        dpkg -i external/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
        dpkg -i external/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
        rm external/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
        rm external/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
```
You can check this here https://github.com/e-m-b-a/emba/blob/b7429b57a6e31a941b79ef4bf1d336d2dba8bab2/installer/ID1_ubuntu_os.sh#L42

torabi12, 2022-09-09:
I followed yor commands above, I reisntalled redis-server with apt, right now I am able to start redis and mongod, but the cve-search is not working.

[image]

[image]

[image]

Busybox test:
[image]

m-1-k-3, 2022-09-09:
That looks not too bad. Mongod and redis server is back up again :)

Now, check the configuration files:
* /etc/mongod.conf
* /etc/redis/redis.conf
This needs to be aligned with the settings in external/cve-search/etc/configuration.ini

Please check the requirepass value in the redis config and align it with the Password value from external/cve-search/etc/configuration.ini
Same with the Host and Port values in mongod.conf

torabi12, 2022-09-09:
The /etc/redis folder is not accessible, even as sudo but it is working again:
[image]

Thank you very much, I don't know what caused this issue.

m-1-k-3, 2022-09-09:
perfect! If you find the reason for this issue please keep us updated.

closed 2022-09-09 by m-1-k-3

Darren-zl, 2023-05-10:
> perfect! If you find the reason for this issue please keep us updated.

I follow the above operations to do, are normal. But I still show cve-search -no ok. I don't know why.
[image]
[image]
[image]
[image]

m-1-k-3, 2023-05-10:
EMBA is doing a testquery to cve-search with busybox. If there are no results EMBA will show that the database is not ready. So, this means you need to populate your database. See also here https://github.com/e-m-b-a/emba/issues/187