e-m-b-a/emba #303: password hashes not being cracked

n0x08, 2022-09-09:
**Describe the bug**
It appears that John the Ripper doesn't like the format of the stored password hashes file 's109_jtr_local_pw_cracking/jtr_hashes.txt' - it exits with this message (from the web report):

[+] Cracking identified password hashes

[*] Starting jtr for the following hashes (runtime: 60m):
servkey.pem:-----BEGINRSAPRIVATEKEY-----Proc-Type:4,
passwd_nfs:admin:yiVXjXdLpGfug:0:0:admin:/:/bin/sh
passwd:root::0:0:root:/root:/bin/sh
passwdr:root:ToCOv8qxP13qs:0:0:root:/root/:/bin/sh
passwd_nfs:root:yiNNyNaXWRwx.:0:0:root:/root/:/bin/sh

**No password hashes loaded (see FAQ)**

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation
2. Analyze firmware which contains password hashes
3. Start EMBA with the following parameters: sudo ./emba.sh -f ~/firmware.dav -l ~/firmware -p ./scan-profiles/default-scan.emba
4. Wait for it to run
5. Browse to html-report//s109_jtr_local_pw_cracking.html of web report

**Expected behavior**
I believe the problem is the jtr_hashes.txt format; when I try to process it manually I get the same error:

nate@emba:~/firmware$ john s109_jtr_local_pw_cracking.txt
No password hashes loaded (see FAQ)

This message is documented at the John website: https://www.openwall.com/john/doc/FAQ.shtml

When I remove the fields before the username (passwd_nfs, passwd, passwdr in the example above) John is able to process it:

nate@emba:~/firmware$ cat testpw
admin:yiVXjXdLpGfug:0:0:admin:/:/bin/sh
root:ToCOv8qxP13qs:0:0:root:/root/:/bin/sh
root:yiNNyNaXWRwx.:0:0:root:/root/:/bin/sh

nate@emba:~/firmware$ john testpw
Loaded 3 password hashes with 2 different salts (descrypt, traditional crypt(3) [DES 128/128 SSE2-16])
Will run 12 OpenMP threads
Press 'q' or Ctrl-C to abort, almost any other key for status
12345            (admin)
Warning: MaxLen = 13 is too large for the current hash type, reduced to 8
duhao            (root)
2g 0:00:00:47 3/3 0.04255g/s 24231Kp/s 24667Kc/s 24668KC/s tj1tl2k..tjkly20
Use the "--show" option to display all of the cracked passwords reliably
Session aborted

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Additional context**
Add any other context about the problem here.

m-1-k-3, 2022-09-09:
Looks like we need some further sanity checks before throwing jtr on it. 
Should be here https://github.com/e-m-b-a/emba/blob/b7429b57a6e31a941b79ef4bf1d336d2dba8bab2/modules/S109_jtr_local_pw_cracking.sh#L46

n0x08, 2022-09-09:
[s108_stacs_password_search.csv](https://github.com/e-m-b-a/emba/files/9538885/s108_stacs_password_search.csv)

closed 2022-09-22 by m-1-k-3