e-m-b-a/emba #308: Where does `config/msf_cve-db.txt` come from.

maringuu, 2022-09-20:
Sorry for the noise, I didn't find any contact channel other than github issues.

I'm trying to perform a cve to exploit mapping. Emba does this pretty nice and I was looking for where the database comes from.

One part of the query is looking at `msf_cve-db.txt`.
I wasn't able to find out how this file is generated/copied from by googeling and looking at the `git log`.

Can you give me some hints? Thank you!

m-1-k-3, 2022-09-20:
You can find the code here: https://github.com/e-m-b-a/emba/blob/b7429b57a6e31a941b79ef4bf1d336d2dba8bab2/helpers/helpers_emba_prepare.sh#L398

If you are running in default mode the regular updates from your git environment will update this datafile for you.

maringuu, 2022-09-20:
Thank you for the answer!
I missed that part when grepping through the code :)

closed 2022-09-20 by maringuu