e-m-b-a/emba #327: Issue 187 CVE search not working

Hellworold, 2022-09-29:
Hi,
When iam trying to run the utility it gives an error CVE- search not working , check issue number 187 thus iam here .
Now i have checked all the steps above.
for some reason this is the output
i have checked with some older versions

I have followed along with issue 187, 
but there is some error in my output sharing the same.

[image]

m-1-k-3, 2022-09-30:
Looks quite good. Probably something went wrong during populating the database. Could you try it again with the following commands:

```
./sbin/db_mgmt_cpe_dictionary.py -p
./sbin/db_mgmt_json.py -p
./sbin/db_updater.py -c
```

BenediktMKuehne, 2022-10-13:
The dependency warning should disappear after: `pip install -U requests`

m-1-k-3, 2022-10-14:
I will include this note in the main CVE-search issue (https://github.com/e-m-b-a/emba/issues/187). Thank you