e-m-b-a/emba #331: Config analysis for more services

m-1-k-3, 2022-10-03:
**Is your feature request related to a problem? Please describe.**
There are multiple services we found in firmware:

- [ ] Apache
- [ ] lighttpd
- [ ] mysql
- [ ] postgresql

**Describe the solution you'd like**
EMBA should be able to analyse the configuration files of these services

m-1-k-3, 2022-10-07:
Regarding _lighttpd_ we could easily check for some of the documented options: https://wiki.alpinelinux.org/wiki/Lighttpd_Advanced_security

```
ssl.cipher-list includes the entries: !aNULL:!eNULL:!EXPORT:!DES:!RC4:!3DES:!MD5:!PSK

The following options are enabled:
ssl.engine    = "enable"
ssl.disable-client-renegotiation = "enable"
ssl.honor-cipher-order = "enable"

ssl.pemfile   is set to something e.g. "/etc/lighttpd/server.pem"
and the permissions of the pem file are strict: chmod 400 /etc/lighttpd/server.pem
```

Further ideas?

gstrauss, 2023-10-02:
FYI: lighttpd 1.4.68 and later TLS defaults are now stronger.  In the modern lighttpd config, the default set of ciphers all support perfect forward secrecy (PFS) and none of the ciphers in the default ciphers list are vulnerable to BEAST or Poodle, so lighttpd recommends leaving other TLS settings at their defaults, too.  `ssl.disable-client-renegotiation = "enable"` is the default (disable client renegotation with TLS 1.2), and `ssl.honor-cipher-order = "enable"` **is not** the default since it is safe to honor the client preference *from within the stronger lighttpd TLS default cipher list*.  Honoring client preference is useful for mobile devices which might not have AES crypto hardware support built-in to the silicon, and so might prefer CHACHA over AES.

PR #1133 (2024-04-18): S36 updates, l10 fixes
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see https://github.com/e-m-b-a/emba/issues/331#issuecomment-1742499493
some little issues in l10

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #331 
fixed l10 issues

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**:
closed 2024-04-18 by HoxhaEndri