e-m-b-a/emba #338: Docker-setup

BenediktMKuehne, 2022-10-06:
Since there are different ways docker is integrated with the WSL there should be an option to skip the Docker-setup.
(docker.io etc)
Other option I would consider is just removing it from the default install.
Currently both EMBA and EMBArk don't work properly inside WSL2-ubuntu with docker-desktop integration.

See here:[WSL2-doc](https://docs.docker.com/desktop/windows/wsl/#best-practices)

Related: [EMBArk-PR-Tracking](https://github.com/e-m-b-a/embark/pull/58)

BenediktMKuehne, 2022-10-06:
"To avoid any potential conflicts with using WSL 2 on Docker Desktop, you must [uninstall any previous versions of Docker Engine](https://docs.docker.com/engine/install/ubuntu/#uninstall-docker-engine) and CLI installed directly through Linux distributions before installing Docker Desktop." - [WSL2-doc](https://docs.docker.com/desktop/windows/wsl/#best-practices)

m-1-k-3, 2022-10-06:
Are you talking about Docker Desktop and not about a plain Ubuntu/Kali with an installation of the docker service?

BenediktMKuehne, 2022-10-06:
It's an issue when the Docker engine is running on the Windows host and WSL provides the docker backend.

[image]

m-1-k-3, 2022-10-06:
hmmm, currently we have only tested and supported the installation of docker within the wsl2 Ubuntu system. I have never tested it with Docker desktop.

See also here: https://github.com/e-m-b-a/emba/wiki/Experimental-WSL-support

Probably we need a different installation path for getting this up and running

BenediktMKuehne, 2022-10-06:
Would require moving CVE-search inside EMBA container.

closed 2022-10-06 by BenediktMKuehne

m-1-k-3, 2022-10-06:
To make this happen we need to find a way to update cve search inside of the container easily