e-m-b-a/emba #342: Installation fails on Kali WSL environment - Error detected - status code 1

rancho666, 2022-10-13:
**The error**
Error detected - status code 1
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer.sh, line 196
Stack Trace:
    [1] main(): ./installer.sh, line 196 -> main -d

Maybe the output of the docker compose doesn't like it?

**To Reproduce**
EMBA installation (default mode) > "sudo ./installer.sh -d"

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: [Kali Linux 2022.03 on WSL W11]

m-1-k-3, 2022-10-13:
Just as a quick note: WSL support is only very experimental. I'm not surprised it fails ;) 
Have you checked our wiki entry on this topic? https://github.com/e-m-b-a/emba/wiki/Experimental-WSL-support

m-1-k-3, 2022-10-13:
I'm wondering on the output of your screenshot. in line 196 should be a command `docker-compose` but the shown error indicates it is `docker compose`

m-1-k-3, 2022-10-14:
uh, I can see the problem. This is the output on my machine:
```
docker-compose version 1.29.2, build unknown
```
And so the check is failing. As quick workaround just comment the docker-compse check (line 196-202). I will fix this issue soon.

rancho666, 2022-10-14:
Workaround successful. Thanks for your help/time :)

closed 2022-10-14 by rancho666

m-1-k-3, 2022-10-15:
@rancho666 could you please check PR https://github.com/e-m-b-a/emba/pull/347
This should fix the issue