e-m-b-a/emba #350: Error detected - status code 127

hikmat30ce, 2022-10-15:
**Describe the bug**
After running in ubuntu docker, I got below error
Error detected - status code 127
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer/I01_default_apps_host.sh, line 51
Stack Trace:
    [1] I01_default_apps_host(): ./installer/I01_default_apps_host.sh, line 51 -> I01_default_apps_host
    [2] main(): ./installer.sh, line 195 -> main -d

**To Reproduce**
Steps to reproduce the behavior:
1. User docker for desktop in mac book pro M1
2. Used below docker file to build image (this is needed to enable ssh from host machine)
      ```
      FROM ubuntu:latest
      RUN apt update && apt install  openssh-server sudo -y
      RUN useradd -rm -d /home/ubuntu -s /bin/bash -g root -G sudo -u 1000 test 
      RUN  echo 'test:test' | chpasswd
      RUN service ssh start
      EXPOSE 22
      CMD ["/usr/sbin/sshd","-D"]
      ```

3. executed below commands 
`git clone https://github.com/e-m-b-a/emba.git
cd emba`
`sudo ./installer.sh -d`

**Expected behavior**
The error should not be thrown

m-1-k-3, 2022-10-16:
just to ensure I understand your setup correctly. You have "docker for desktop" on your mac and you install Ubuntu in there and then EMBA in default mode within this dockerized Ubuntu?

m-1-k-3, 2022-10-16:
The error is from updating the pip requests package without installing pip ... could you please check the PR. Probably this will fix this error. We will see if you are running into further issues with your setup.

m-1-k-3, 2022-10-17:
Reopen if not fixed

closed 2022-10-17 by m-1-k-3