e-m-b-a/emba #353: Pause and resume functionality

a-huk, 2022-10-18:
This idea came as I was running a large firmware (4gb) analysis and it is taking relatively long (40 hours and counting) on relatively modern and powerful hardware. No issues there, but it would be nice if the analysis could be paused and resumed at a later date. I understand that some checks and modules can not be resumed in the middle of them, but simply restarting from the last completed module would be helpful. 

Thanks for the awesome project

m-1-k-3, 2022-10-18:
You could try to stop the analysis via <Strg>+<C>. If you start the same firmware test again EMBA will give you a note that there is a test already available and not finished. EMBA will then try to resume this scan. This works most of the time not too bad but has currently some limitations:
* Web reporter looses links
* Live tester modules (-Q) are currently not supported to be resumed
* Probably a lot more limitations and bugs
* Warning: this feature is highly experimental!

m-1-k-3, 2022-10-18:
Should look like the following screenshot:

[image]

press "n" and EMBA will try to use as many results from the first scan as possible

m-1-k-3, 2022-11-04:
Can we get an update on this issue?

closed 2023-02-28 by m-1-k-3