e-m-b-a/emba #356: cve-search formatting

a-huk, 2022-10-19:
**Describe the bug**
When scanning firmware with the default options any time there is a cve-search then the output is weird and seems to be always tabbed. I suppose that this is not intended.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the default scan command
2. Observe

**Expected behavior**
I would expect that every cve-search - testing & cve-search - ok would have their own line. It might also be nice to have some counter/progress bar. A tqdm equivalent would be nice or to be honest just a simple counter 5/1650 would suffice.

**Screenshots**
[image]
This is with yeara but the same occurs for example with grepit
**Desktop (please complete the following information):**
 - OS: Kali Linux WSLg

m-1-k-3, 2022-10-19:
Yea, the output is really not nice. Are you running your scan in threading mode? I think not because then you will see a lot of dots. So, your scan will probably take forever ;) Try -t or the default profile `-p ./scan-profiles/default-scan.emba` or the full-scan profile.

Additionally we have started with a status bar a while ago. You can enable it via `-B`. But we warned the bar will get destroyed on different firmwares. It is currently in an unstable state and need some more attention :)

a-huk, 2022-10-19:
Indeed I am not running with threading, I don't remember if I added a profile option or just run it without one.

I assume that threading messes up the output of the terminal but the final report should be fine, no?
Second, I would like to contribute to the project, is there anything that can be implemented or am I free to change things and open pull requests? Lastly would you allow other languages that bash or do you want to keep it bash only?

m-1-k-3, 2022-10-19:
> Indeed I am not running with threading, I don't remember if I added a profile option or just run it without one.

If you have some decent power in your environment EMBA will use it and speedup things massively :)

> I assume that threading messes up the output of the terminal but the final report should be fine, no? 

Multiple modules are going to report into one interface ... so, the output is very confusing ;) The final report is fine. If not it is a bug ;)

> Second, I would like to contribute to the project, is there anything that can be implemented or am I free to change things and open pull requests? 

Awesome ... you are welcome! I would recommend checking the issues if there is something you are interested in. Then, there is some further information in [CONTRIBUTORS.md](https://github.com/e-m-b-a/emba/blob/master/CONTRIBUTORS.md) and [CONTRIBUTING.md](https://github.com/e-m-b-a/emba/blob/master/CONTRIBUTING.md). If you have other ideas we can discuss them as issues or in the discussion section. You are also welcome in just opening a PR for fixing things. Ensure to  use our [check script ](https://github.com/e-m-b-a/emba/blob/master/check_project.sh) before a PR.

> Lastly would you allow other languages that bash or do you want to keep it bash only?

Bash only for EMBA.

m-1-k-3, 2022-10-19:
btw. if you prefer python we have EMBArk as web frontend for EMBA see [here](https://github.com/e-m-b-a/embark)

a-huk, 2022-10-20:
Btw how long is the yala scan supposed to run? I have no multi-threading enabled, and have some 1700 files unpacked (4gb total). The yala part has been running for 48h. Can I maybe skip it?

m-1-k-3, 2022-10-20:
You mean the Yara module? If you are running in Single Threading mode (without -t parameter or with no profile activated) it will take ages as it is running only on one core of these 12 ;) I do not think this is what you want. Stop the scan and restart it with the default profile or at least with `-t `enabled.

The Yara module is one of the longer running modules. If you want to deactivate it globally for EMBA you can include it into `config/module_blacklist.txt` ... see also this wiki entry: https://github.com/e-m-b-a/emba/wiki/Tweak-your-scan

m-1-k-3, 2022-10-24:
@a-huk please check the modifications. I hope it should be cleaner now.