e-m-b-a/emba #362: emba bug

kashifnaeem745, 2022-10-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: <Insert URL>
3. Start EMBA with the following parameters: sudo ./emba.sh <insert further parameters>
4. additional steps
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2022.03
- EMBA version: v1.1.2 or current master branch
- Installation method: default with up to date docker image

**Additional context**
Add any other context about the problem here.

m-1-k-3, 2022-10-26:
Any progress?

m-1-k-3, 2022-11-12:
no further response - closed

closed 2022-11-12 by m-1-k-3