e-m-b-a/emba #367: System Mode / User Mode Emulation

keeplearning-rgb, 2022-10-26:
I just need to emulate my firmware image using emba without any analysis. Though EMBA do provide a directory to re run the emulation later but is there any way so that i can just create the emulation both for user mode / System mode without doing any analysis of the vulnerabilities

m-1-k-3, 2022-10-27:
You should be able to only run all pre-checker modules, the usermode emulator (s115) and the initial system-mode emulator (l10), skipping the finalizers with this ugly trick ;)

```
└─$ sudo ./emba.sh -f ~/firmware.bin -l ~/firmware-log -p ./scan-profiles/default-scan-emulation.emba -m s115 -m l10 -m f199
```

f199 is not available and so you are skipping all f modules.

Give it a try.

keeplearning-rgb, 2022-10-27:
If you are aware is emba supporting armel architecture for system emulation as of now??

m-1-k-3, 2022-11-04:
Can we get an update on this issue?

keeplearning-rgb, 2022-11-04:
This issue is resolved.  I am able to run the emulation only without any analysis by providing f199 in the command line as specified by you above.
Thank you

closed 2022-11-04 by m-1-k-3