e-m-b-a/emba #370: System Mode Emulation stuck in infinite loop

keeplearning-rgb, 2022-10-27:
I Tried to do the system mode emulation for a mipsel architecture firmware image but output is stuck at infinite loop.
Don't know the exact reason.

Attaching the log file for the same 
This is the main error 
[image]

Attached full logs as well for more details
[qemu.initial.serial_squashfs-root_mipsel-28648.log](https://github.com/e-m-b-a/emba/files/9881974/qemu.initial.serial_squashfs-root_mipsel-28648.log)

m-1-k-3, 2022-10-27:
Could you provide a download link to the firmware or the firmware itself offline?

m-1-k-3, 2022-10-27:
The log is from the network detection process and should be stopped after a while. As every process runs into a kernel exception this probably needs further debugging.

m-1-k-3, 2022-10-28:
Let EMBA finish the job ... 
[image]

keeplearning-rgb, 2022-10-28:
How much time it took for you, because for me i tried re-running it several times but never reached to this output.

sudo ./emba.sh -f /root/Desktop/P3225-LVE_Mk_II_9_80_3_13.bin -l ~/firmware-log -p ./scan-profiles/default-scan-emulation.emba -m s115 -m l10 -m f199
You just ran it with this command, and it worked for you?

Because this is not working for me.
Am i missing something in this?

m-1-k-3, 2022-10-28:
hmmm ... interesting. I stopped the test after the ping succeeded (~35min). This was my command line:

```
└─$ sudo ./emba.sh -f ~/Axis/P3225-LVE_Mk_II_9_80_3_14.bin -l ~/emba_logs_axis -S -p ./scan-profiles/default-scan-emulation.emba -m s199 -m l10 -m f199
```
Are you running on a Kali or Ubuntu host? WSL? Default or full mode?

keeplearning-rgb, 2022-10-28:
I am running it on Ubuntu. 
It is a separate machine.

I have initiated the emulation once again. 
Will ping you with the results in another 30 mins.

Any comment on ARMEL emulation part??

m-1-k-3, 2022-10-28:
With the -S you are running in STRICT mode. If EMBA is running into issues there should be an error log file in the log directory.

> Any comment on ARMEL emulation part??

ARMel should work

keeplearning-rgb, 2022-10-28:
Reached to the same output. But now i have concern regarding the accessibility of the emulated environment.

The ip address which is mentioned as inet address if we access it over browser, it should show us the emulated firmware OR if not what's the purpose of that.

Second after this analysis is complete, we get a directory with files to re emulate the system emulation whenever required.
But its not working for me, its just stuck at 
[image]

My end goal to emulate the firmware is to do the dynamic analysis on the same.
Also to to do the false positive analysis of the vulnerabilities identified my emba.

m-1-k-3, 2022-10-29:
Perfect. This means EMBA was able to get it up and running and the network interface was configured successfull to reach the system via ICMP. As it was not possible to get some services up and running it is now your job to go the next mile. EMBA gave you a solid base :)

Regarding the error you need to install the needed packages on your host. First package is `uml-utilities`. For further packages you probably need take a look at the system mode installer in `installer/IL10_system_emulator.sh`

keeplearning-rgb, 2022-11-03:
While analyzing the html report for system emulation. I got that emba is facing some kernel panic issues which i was facing while trying to emulate the firmware with firmadyne OR FirmAE. Do emba stores complete logs of what error faced and how it bypassed the same. Those can be really helpful to debug different emulations.

[image]

m-1-k-3, 2022-11-03:
In the last line of your output EMBA says what she found and what is the next step:
```
Warning: Kernel panic with failed rdinit found - testing init
```

keeplearning-rgb, 2022-11-03:
I have another firmware file with some doubts, not able to find the download link. Can i share it offline to you?

m-1-k-3, 2022-11-03:
You can upload it somewhere and send me the link via a PM on [twitter](https://twitter.com/securefirmware)

keeplearning-rgb, 2022-11-03:
I don't think i can do PM to this on twitter. Tried but no luck

m-1-k-3, 2022-11-03:
Can you just share the link here? So, others are also able to check it and help

keeplearning-rgb, 2022-11-03:
https://drive.google.com/file/d/1TBhk4RqKc9CXtqPHZsG3cLdlqQejP2pF/view?usp=sharing

Try this

m-1-k-3, 2022-11-03:
looks interesting but broken:

```
└─$ sudo hexdump /emba_logs_Avg-HDH264/firmware/patool_extraction/ubifs_u_ubi_extracted/152354692/rootfs/bin/busybox
0000000 0000 0000 0000 0000 0000 0000 0000 0000
*
01ae4a0
```

keeplearning-rgb, 2022-11-03:
_Avg-HDH264-FW-t210.bin.extracted/_ubifs_u.extracted/ubifs-root/152354692/rootfs

You can try extracting the filesystem using -Me option in binwalk and the above path will give you the root directory.
I think emba does it automatically. Didn't faced any error while extracting in emba.

Emulation is not happening, but as per your comment armel is supported but emba logs shows not supported

m-1-k-3, 2022-11-03:
all of the binaries are broken in automatic extraction:

```
└─$ hexdump -C /home/m1k3/firmware-stuff/emba_logs_Avg-HDH264/firmware/patool_extraction/ubifs_u_ubi_extracted/152354692/rootfs/sbin/ctrlaltdel
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000015a0  00 00 00 00 00 00 00 00  00 00 00 00              |............|
000015ac
```

The analysis of this firmware does not really make sense. Probably something during UBI extraction was going wrong or the complete firmware update is corrupted.

m-1-k-3, 2022-11-03:
Update: via manual UBI extraction it is possible to get the correct root filesystem:

```
─$ file _ubifs_u_manual_extracted/152354692/rootfs/bin/busybox
_ubifs_u_manual_extracted/152354692/rootfs/bin/busybox: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, for GNU/Linux 2.6.16, BuildID[sha1]=1f5b98b02eeabf92a062c791f0a7599a37779700, stripped
```

You can do it the following way:

```
└─$ binwalk -e Avg-HDH264-FW-t210.bin    
└─$ cd _Avg-HDH264-FW-t210.bin.extracted                             
└─$ ubireader_extract_files -l -i -w -v -o _ubifs_u_manual_extracted ubifs_u
```

Afterwards you should be able to run EMBA on this extracted UBI filesystem:

``` 
└─$ sudo ./emba.sh -f ~/Downloads/_Avg-HDH264-FW-t210.bin.extracted/_ubifs_u_manual_extracted -l ~/firmware-stuff/emba_logs_Avg-HDH264 -S -p ./scan-profiles/default-scan-emulation.emba 
```

I need to take a deeper look why EMBA is breaking the UBI filesystem during automated extraction.

keeplearning-rgb, 2022-11-03:
Are you able to do the emulation on the extracted file system??

m-1-k-3, 2022-11-03:
[image]

System boots up and is reachable via ICMP

keeplearning-rgb, 2022-11-07:
I am able to do the system emulation for the above provided framework.
Thanks a lot

I have one generic doubt, if you have any idea on that kindly help.
I was looking at QEMU logs and saw that at multiple places we get permission denied error while it tries to get some services up. Can you refer to something which can help me solve this thing and what's the reason that we are facing such errors.

[image]

m-1-k-3, 2022-11-07:
We are going to improve this in https://github.com/e-m-b-a/emba/pull/380

closed 2022-11-09 by BenediktMKuehne