e-m-b-a/emba #371: Different results for different firmware files

keeplearning-rgb, 2022-10-27:
I tried analysis using emba on two different firmware files.
But for the one with mipsel architecture there are more thorough results while in case of armel architecture only vulnerabilities identified are with the kernel version.

I am aware that for emulation part it doesn't support armel but does the overall analysis of emba depends on what architecture a particular file has??

m-1-k-3, 2022-10-27:
are both of them the same firmware for different architectures? Otherwise I think you are just testing different firmwares and so you will get different results. Could you provide the firmware for testing?

m-1-k-3, 2022-11-04:
Can we get an update on this issue?

keeplearning-rgb, 2022-11-04:
I have already provided you the different firmwares which i was testing in the separate thread.  This issue can be closed. I can ask further queries in the other threads.
Thank you

closed 2022-11-04 by m-1-k-3