e-m-b-a/emba #375: UBI extraction corrupts ELF files

m-1-k-3, 2022-11-03:
See https://github.com/e-m-b-a/emba/issues/370#issuecomment-1301821783

Testfirmware: https://drive.google.com/file/d/1TBhk4RqKc9CXtqPHZsG3cLdlqQejP2pF/view?usp=sharing

EMBA corrupts ELF files during extraction. Via manual extraction the root filesytem is not corrupted!

```
─$ file _ubifs_u_manual_extracted/152354692/rootfs/bin/busybox
_ubifs_u_manual_extracted/152354692/rootfs/bin/busybox: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, for GNU/Linux 2.6.16, BuildID[sha1]=1f5b98b02eeabf92a062c791f0a7599a37779700, stripped
```

You can do it the following way:

```
└─$ binwalk -e Avg-HDH264-FW-t210.bin    
└─$ cd _Avg-HDH264-FW-t210.bin.extracted                             
└─$ ubireader_extract_files -l -i -w -v -o _ubifs_u_manual_extracted ubifs_u
```

Afterwards you should be able to run EMBA on this extracted UBI filesystem:

``` 
└─$ sudo ./emba.sh -f ~/Downloads/_Avg-HDH264-FW-t210.bin.extracted/_ubifs_u_manual_extracted -l ~/firmware-stuff/emba_logs_Avg-HDH264 -S -p ./scan-profiles/default-scan-emulation.emba 
```

I need to take a deeper look why EMBA is breaking the UBI filesystem during automated extraction.

m-1-k-3, 2022-11-03:
looks like our lzo installation from the docker container is somehow corrupt:

```
decompress Warn: LZO Error: module 'lzo' has no attribute 'decompress'
_process_reg_file Warn: inode num:157 :can't concat NoneType to bytearray
```
Probably this is the source of the shown issues in extracting ubis

keeplearning-rgb, 2022-11-07:
Manually extracting the files using ubi_extractor works for me as well

m-1-k-3, 2022-11-07:
We are going to fix this in the container soon

closed 2022-11-08 by BenediktMKuehne

m-1-k-3, 2022-11-18:
The updated docker image should now be able to handle ubi extraction again.
Thank you for your patience.