e-m-b-a/emba #377: Analysis during system mode emulation

keeplearning-rgb, 2022-11-04:
During the system mode emulation are we covering the scanning part as mentioned in last point of system mode emulation wiki.

[image]

m-1-k-3, 2022-11-04:
If you run it with -Q parameter or with the default-emulation or with full-scan profile AND we have services up and running it should run automatically.

keeplearning-rgb, 2022-11-04:
Which services you are referring to here?

m-1-k-3, 2022-11-04:
If we detect a service up and running in L10 further Lxx modules are automatically running. Does not matter which service. This should be indicated with the output of "[+] Network services are available."

m-1-k-3, 2022-11-04:
[image]

keeplearning-rgb, 2022-11-07:
Understood. It works for me in one of the firmware where some services got up and running. 
Nmap scan was done for those.

Thank you. this issue can be closed

closed 2022-11-07 by m-1-k-3