e-m-b-a/emba #378: error no more space.... but plenty of diskspace available

rjgjansen, 2022-11-04:
following the installation instructions  https://github.com/e-m-b-a/emba
the installation fails 
[image]

there is available disk space
[image]

what am i doing wrong

m-1-k-3, 2022-11-04:
The docker images needs around 12-13gigs and in a default installation the docker directory is on / which has only 8.6gigs. You can try to move your docker directory to another disk and make a sym link to it. Usually this directory is `/var/lib/docker`.

closed 2022-11-11 by BenediktMKuehne

rjgjansen, 2022-11-15:
HI,

tnx...  a fellow student claimed it was a memory problem....  i put an extra 8 gb of mem from an other computer and the problem was gone.  also the other error messages were gone.