e-m-b-a/emba #382: Getting interactive shell while re-running the System Emulation

keeplearning-rgb, 2022-11-09:
While i was trying to emulate the firmware with firmadyne i saw if emulation is successful, it provides us an interactive shell.
On the other hand EMBA resolves many problems faced in firmadyne to emulate a firmware it doesn't provided an interactive shell upon successful emulation.

Can this feature be included in upcoming releases? Because from my point of view getting an interactive shell upon successful emulation provided us a better hold in debugging and interacting with the emulated firmware

m-1-k-3, 2022-11-09:
This is already addressed in https://github.com/e-m-b-a/emba/pull/380
In the future you can re-start your emulation with ./run.sh and connect via telnet to port 4321 to access the system

keeplearning-rgb, 2022-11-09:
Getting connection refused error 

[image]

m-1-k-3, 2022-11-09:
have you already applied the PR?
check your run.sh script for an entry `-serial telnet:localhost:4321,server,nowait`

You can also add it manually and then you need to `telnet localhost 4321`

keeplearning-rgb, 2022-11-09:
Added the telnet parameter but same error.
This is the nmap output which shows telnet port is open 

[image]

keeplearning-rgb, 2022-11-09:
Its connected finally but no shell still.
[image]

m-1-k-3, 2022-11-09:
So, there is no shell listening in this firmware. Test it with a known working firmware and you should have access to the cli.