e-m-b-a/emba #386: Looking for MIPS64 firmware

m-1-k-3, 2022-11-11:
**Is your feature request related to a problem? Please describe.**

We are looking for special architecture firmware for further testing and improving EMBA

**Describe the solution you'd like**

If you are aware of devices running **Linux** based **MIPS64 eb/el** firmware get in touch with us or just post into this issue.

jblu42, 2022-11-14:
@m-1-k-3: A Good way to start for all the "Looking for firmware" stuff is the table of OpenWRT, there you can find Wifi Routers by Architecture and try their original firmware or maybe even OpenWRT:
https://openwrt.org/docs/techref/instructionset/start

m-1-k-3, 2022-11-18:
Another MIPS64 firmware found in the Qemu Advent Calendar: https://www.qemu-advent-calendar.org/2018/download/day22.tar.xz

jblu42, 2022-11-18:
According to OpenWRT Table e.g. Ubiquiti EdgeRouter ER-8
https://www.ui.com/download/edgemax/#
https://dl.ui.com/firmwares/edgemax/v2.0.9-hotfix.4/ER-e200.v2.0.9-hotfix.4.5521907.tar

m-1-k-3, 2022-11-18:
Found that. But I'm looking for further firmware images for testing different cases

m-1-k-3, 2023-11-20:
https://github.com/e-m-b-a/emba/issues/897

closed 2023-11-20 by m-1-k-3