e-m-b-a/emba #390: Looking for ARM64 firmware

m-1-k-3, 2022-11-12:
**Is your feature request related to a problem? Please describe.**

We are looking for special architecture firmware for further testing and improving EMBA

**Describe the solution you'd like**

If you are aware of devices running **Linux** based **ARM64** firmware get in touch with us or just post into this issue.

torabi12, 2023-01-19:
I checked some legacy Mikrotik router firmware files with emba. There was no issue during the scan and the reports were also fine. I ust wanted to mention this:
https://download.mikrotik.com/routeros/7.2/routeros-7.2-arm64.npk

was successfully tested with emba.

Other versions and architectures can be found here:
https://mikrotik.com/download/archive

Test time was about 26 minutes, which is not short but not so long.

[image]

m-1-k-3, 2023-01-20:
Very interesting firmware. Thank you for pointing on this beauty :)

m-1-k-3, 2023-11-20:
https://github.com/e-m-b-a/emba/issues/897

closed 2023-11-20 by m-1-k-3