e-m-b-a/emba #391: p60 running, no deep extraction set

jblu42, 2022-11-14:
Hi. I am trying to analyse parts of a firmware that I have packed into a single tar file. The tar file does include only parts the filesystem of a device because I have several tar files for different partitions on the device.

P59 correctly extracts the filesystem, but then p60_firmware_bin_extractor tries to extract linux library files and ends up with some 10k files that are no longer intact.

I use the default emba settings with full scan profile, which should not do deep extraction.

Edit: While writing this I noticed in the logs, that probably because no root filesystem is found, emba tries to deep extract the files.

Any suggestion how to disable this behaviour or a hint how emba detects if there is a root filesystem present?

m-1-k-3, 2022-11-14:
You can find the detection of the root filesystem here https://github.com/e-m-b-a/emba/blob/20ca0241dfe678f3da68763069dc3284c3a4fc52/helpers/helpers_emba_prepare.sh#L412

m-1-k-3, 2022-11-15:
You should be able to disable it with the module blacklist: https://github.com/e-m-b-a/emba/wiki/Tweak-your-scan#blacklist-modules

jlucius, 2022-11-16:
@m-1-k-3 Still the user that opened this ticket, for company process reasons it takes a looong time to allow official contributions and you have to jump through a lot of hoops to get there, so this is private work in private time. - end disclaimer :-)

After looking at the rootfs detection function, the filesystem in question here fails all checks, but I did see as some last resort you check where the bash is located. Unfortunately the android system also has no bash, but it has the /bin/sh as shell.
I would propose adding a rootfs check for /bin/sh, this would solve this case.
(Pull request follows)

jblu42, 2022-11-21:
Works with provided patch

closed 2022-11-21 by jblu42